OpenC3/cosmos #814: cosmos-project-openc3-cosmos-init looping restart when volumes exist

skliper, 2023-09-08:
**Describe the bug**
First `./openc3.sh start` works (fresh w/ no volumes).  Any start after that results in cosmos-project-openc3-cosmos-init looping with the following error:
```
mc: <ERROR> Unable to make user/group policy association. The specified policy change is already in effect. (Specified policy update has no net effect).
```

**To Reproduce**
```
./openc3.sh start
# wait for start
./openc3.sh stop
# wait for stop
./openc3.sh start
# observe init container looping every few seconds
```
 
**Expected behavior**
Clean start when volumes exist.  Side note, there's an anonymous volume that gets created every start.  Is this really intended?  Ends up cluttering docker volumes up, and they don't get cleaned w/ `./openc3.sh cleanup`.

**Environment (please complete the following information):**
- OS: Windows 10
- OpenC3 COSMOS Version: 5.10.1 (Didn't see this in 5.10.0)
- Browser: NA

ryanmelt, 2023-09-08:
This was closed by #789 
Will be fixed in the 5.11.0 release next week.
Till then recommend stopping the init container, and commenting out of compose.yaml after a first start.

I'm not aware of the anonymous volume.  Leaving this ticket open until that is investigated.

skliper, 2023-09-08:
Thanks @ryanmelt, note the anonymous volume shows up as attached to `cosmos-project-openc3-redis-ephemeral-1`.  

I think it's due to not having /data in compose.yaml (like openc3-redis used of the named openc3-redis-v volume) it creates an anonymous one.

skliper, 2023-09-08:
Note I achieved similar behavior (ephemeral volume) by creating/using a named volume openc3-redis-ephemeral-v and adding `docker volume rm openc3-redis-ephemeral-v` at the end of the stop script.

ryanmelt, 2023-09-11:
closed by #815

closed 2023-09-11 by ryanmelt