OpenC3/cosmos #816: How to configure COSMOS to talk to external hardware

ascended121, 2023-09-11:
**Is your feature request related to a problem? Please describe.**
I'm trying to setup COSMOS to talk to an external piece of hardware (which does not reside in docker). I've followed the [Getting Started](https://docs.openc3.com/docs/v5/gettingstarted) section, creating my own plugin with my target definition and my own interface definition, specifying the proper UDP ports for my hardware, however COSMOS never receives any telemetry from my hardware.

The demo defines an interface which communicates with `host.docker.internal` on ports 8080/8081. Since my target is outside of docker, it seems likely that I need to provide a port mapping from the host into the docker environment somewhere, but I'm not sure where that should go. Maybe in [`compose.yml`](https://github.com/OpenC3/cosmos-project/blob/d3b62aea354762100e3bd189ac7a0a5828096723/compose.yaml)??, although even then its not 100% clear which service it should be applied to.

**Describe the solution you'd like**
I think talking to hardware outside of docker is likely the more common use case for docker, so it would be helpful if the tutorial walked through that case, rather than the case where you're talking to a target within docker (which seems like it'd be a niche use case). 

**Additional context**
N/A

ryanmelt, 2023-09-11:
The most common connections are tcp/ip clients which don't require opening ports into Docker, but you are correct we should give better directions for this.

You need to port forward the udp receive port into docker in the compose.yaml file.  The port will need to be opened on the operator container. I'm on my phone now, but will try to get the exact lines later today.

ryanmelt, 2023-09-11:
Need to add these ports lines to your compose.yaml file.

```
  openc3-traefik:
    ports:
      - "8080:8080/udp"
      - "8081:8081/udp"
```

Note you probably don't need both, but I don't know which is the read port.

ascended121, 2023-09-12:
I'm attempting to use port 12345 (arbitrarily chosen). I updated my ports as shown:
```
  openc3-traefik:
    ports:
      - "127.0.0.1:2900:80"
      - "127.0.0.1:2943:443"
      - "12345:12345/udp"
```
(the other entries were already there, so I didn't touch them), restarted:

```
user@cosmos ~/cosmos$ docker-compose restart
[+] Restarting 8/8e restart
 ✔ Container cosmos-openc3-cosmos-cmd-tlm-api-1        Started                                                     2.2s
 ✔ Container cosmos-openc3-traefik-1                   Started                                                     2.6s
 ✔ Container cosmos-openc3-minio-1                     Started                                                     2.5s
 ✔ Container cosmos-openc3-redis-1                     Started                                                     2.3s
 ✔ Container cosmos-openc3-redis-ephemeral-1           Started                                                     2.3s
 ✔ Container cosmos-openc3-cosmos-script-runner-api-1  Sta...                                                      1.9s
 ✔ Container cosmos-openc3-operator-1                  Started                                                     2.4s
 ✔ Container cosmos-openc3-cosmos-init-1               Started                                                     1.4s
```

but I don't see that docker has exposed any additional ports:

```
 docker container ls --format "table {{.ID}}\t{{.Names}}\t{{.Ports}}" -a
CONTAINER ID   NAMES                                      PORTS
f6b0c920efa6   cosmos-openc3-cosmos-init-1
5d101cd83df6   cosmos-openc3-cosmos-script-runner-api-1   2902/tcp
51c349ef8df7   cosmos-openc3-cosmos-cmd-tlm-api-1         2901/tcp
c72828de16dc   cosmos-openc3-operator-1
e6cff5a2731a   cosmos-openc3-traefik-1                    127.0.0.1:2900->80/tcp, 127.0.0.1:2943->443/tcp
cbded71a1cee   cosmos-openc3-minio-1                      9000/tcp
d4e1b680fc0e   cosmos-openc3-redis-1                      3680/tcp, 6379/tcp
b35a667bcc9d   cosmos-openc3-redis-ephemeral-1            3680/tcp, 6379/tcp
```

I can verify that telemetry is getting to the host via tcpdump, but I'm still not seeing telemetry in COSMOS (presumably because docker isn't showing the port as being mapped.

If its relevant, we're currently using COSMOS 5.4.2. I can look into upgrading if its relevant to this problem.

```
$ docker --version
Docker version 20.10.21, build 20.10.21-0ubuntu1~20.04.2
$ docker-compose --version
Docker Compose version v2.15.1
```

ryanmelt, 2023-09-12:
openc3-traefik is the wrong container.  Needs to be the openc3-operator, sorry that was a typo on my part.  Corrected above.

ryanmelt, 2023-09-12:
Reopened to better document this.

ascended121, 2023-09-12:
Made that update

```
  openc3-operator:
    ports:
      - "12345:12345/udp"
 ```

(I had to add the ports section, because there were no ports specified previously), restarted, but I still see that that container has no ports mapped:

```
$ docker container ls --format "table {{.ID}}\t{{.Names}}\t{{.Ports}}" -a
CONTAINER ID   NAMES                                      PORTS
f6b0c920efa6   cosmos-openc3-cosmos-init-1
5d101cd83df6   cosmos-openc3-cosmos-script-runner-api-1   2902/tcp
51c349ef8df7   cosmos-openc3-cosmos-cmd-tlm-api-1         2901/tcp
c72828de16dc   cosmos-openc3-operator-1
e6cff5a2731a   cosmos-openc3-traefik-1                    127.0.0.1:2900->80/tcp, 127.0.0.1:2943->443/tcp
cbded71a1cee   cosmos-openc3-minio-1                      9000/tcp
d4e1b680fc0e   cosmos-openc3-redis-1                      3680/tcp, 6379/tcp
b35a667bcc9d   cosmos-openc3-redis-ephemeral-1            3680/tcp, 6379/tcp
```

ryanmelt, 2023-09-12:
The port shows up for me:

```
ryanmelt@Ryans-MBP cosmos % docker container ls --format "table{{.ID}}\t{{.Names}}\t{{.Ports}}" -a
CONTAINER ID   NAMES                                      PORTS
a39848c0cb36   cosmos-openc3-cosmos-init-1                
7cca8f3a87c1   cosmos-openc3-cosmos-cmd-tlm-api-1         2901/tcp
994cdbc149b7   cosmos-openc3-cosmos-script-runner-api-1   2902/tcp
cd386dca9e5f   cosmos-openc3-operator-1                   0.0.0.0:12345->12345/udp
dd6491cb2659   cosmos-openc3-traefik-1                    127.0.0.1:2900->80/tcp, 127.0.0.1:2943->443/tcp
173e18114381   cosmos-openc3-redis-1                      3680/tcp, 6379/tcp
7f97e94acf2b   cosmos-openc3-minio-1                      9000/tcp
f3e0636c771a   cosmos-openc3-redis-ephemeral-1            3680/tcp, 6379/tcp
dffc2cb27184   agitated_visvesvaraya                      161/udp
```

ryanmelt, 2023-09-12:
Here is my complete operator section:
```
  openc3-operator:
    # For rootless podman - Uncomment this user line and comment out the next
    # user: 0:0
    user: "${OPENC3_USER_ID}:${OPENC3_GROUP_ID}"
    image: "${OPENC3_REGISTRY}/openc3inc/openc3-operator:${OPENC3_TAG}"
    restart: "unless-stopped"
    # ports:
    #  - "127.0.0.1:7779:7779" # Open port for the demo router
    depends_on:
      - "openc3-redis"
      - "openc3-redis-ephemeral"
      - "openc3-minio"
    volumes:
      - "openc3-gems-v:/gems:ro"
      - "./plugins:/plugins:z"
      - "./cacert.pem:/devel/cacert.pem:z"
      # Add access to the entire C drive on Windows
      # - "/c:/c"
    logging:
      driver: "json-file"
      options:
        max-size: "10m"
        max-file: "3"
    environment:
      - "GEM_HOME=/gems"
    env_file:
      - ".env"
    extra_hosts:
      - host.docker.internal:host-gateway
    ports:
      - "12345:12345/udp"
```

ascended121, 2023-09-12:
Apparently `docker-compose restart` will not create new port mappings. Had to `docker compose down` and then `docker compose up` again to get it to take effect.

I now see telemetry. Thanks for the help @ryanmelt !!

FWIW, I do think the documentation would benefit from describing the process of exposing ports, but I'll leave it up to you what to do with this issue.

PR #950 (2023-12-11): Add Key Concepts page
I'll need to regenerate the site before merge but check this out and let me know what you think. This is the result of a few recent discussions we've had about confusion about the architecture and things we're doing.

closes #816
closed 2023-12-11 by jmthomas