OpenC3/cosmos #827: Best way to log CFS event messages

ascended121, 2023-09-13:
CFS uses event messages to provide text messages to operators when key events occur.

While its possible to define a telemetry packet to parse the packet those events are send down in, its hard to monitor them in the COSMOS windows (requiring that one have the Packet Viewer up and set to that packet, and that they don't come in too quickly to read).

I've seen other ground systems log those messages to the terminal, to make them visible to the operator. Ideally I'd like to log them to the scrolling log in the CmdTlmServer. Is there an easy way to do that in COSMOS?

It seems like one might be able to define a `GENERIC_READ_CONVERSION` or `READ_CONVERSION` on the telemetry item containing the message, and instead of converting the value, just log it, but I'm not sure if the logging apis are available in those contexts. Can you advise on the best way to approach this?

If there's a simple solution, it might also make sense to document it in the `COSMOS and NASA cFS` section of the docs.

ryanmelt, 2023-09-13:
This is the primary use case for the Data Viewer tool.  To show a scrolling history of messages.   Does that meet the need?

closed 2023-09-17 by ryanmelt

ascended121, 2023-09-19:
While it is possible to setup the DataViewer to display the packets containing the event messages, it requires either a lot of clicking on the [image] button to step through the timeline, or very precise movement of the scrollbar to view each message in a set that were received at similar times, neither of which is viable for quickly assessing the result of a series of events.

Is it possible to log the event messages in the CmdTlmServer log window?

ryanmelt, 2023-09-19:
Yep.  You could do that from a custom interface, or protocol.
You could also use a Processor, but those currently get executed twice (in the InterfaceMicroservice, and the DecomMicroservice).