OpenC3/cosmos #834: Python API logging connection refused

apogeecmb, 2023-09-18:
Hello,

I've been trying to experiment with the new python scripting API from a separate python script running outside of OpenC3.  I used the cosmos_v5_example.py script in the repo as starting point, but I'm coming up against an error when I try to send a command using the API.  The command is successfully sent to OpenC3, verified by the fact that target responds as expected, but there is a redis.exceptions.ConnectionError when it goes to log the command.  

Looking at the traceback, I was able to track the origin of the exception to openc3/python/openc3/script/commands.py.  Specifically, the exception is triggered in the _cmd method when _log_cmd is called.  I have set my OPENC3_REDIS_EPHEMERAL_HOSTNAME, OPENC3_REDIS_USERNAME, and OPENC3_REDIS_PASSWORD environment variables appropriately to match what is set in the OpenC3 docker configuration.

Any suggestions on what I might still have wrong?  Here's the last part of the traceback.  The connection hostname and port match what I expected based on the hostname I set and the default port.  Since redis is running in a Docker container, I tried a couple different hosts (localhost, 172.17.0.1, etc) without luck.  Do I have the hostname wrong?  Should it be more than just the IP?

```
File "/home/[user]/.local/lib/python3.10/site-packages/redis/connection.py", line 1073, in get_connection

    connection.connect()

  File "/home/[user]/.local/lib/python3.10/site-packages/redis/connection.py", line 265, in connect

    raise ConnectionError(self._error_message(e))

redis.exceptions.ConnectionError: Error 111 connecting to 172.17.0.1:6379. Connection refused.
```
Thanks for any guidance you can provide.

ryanmelt, 2023-09-18:
Set the OPENC3_NO_STORE environment variable to smoething 
export OPENC3_NO_STORE=1

apogeecmb, 2023-09-19:
That resolved my issue.  Thanks!

closed 2023-09-19 by apogeecmb