OpenC3/cosmos #851: Canvas Widget From TLM

tsjordan-eng, 2023-10-02:
Can we make the canvas widget react to telemetry? I've been trying to re-create the [example widget from the COSMOS 4](https://ballaerospace.github.io/cosmos-website/docs/v4/screens#canvasdot) documentation to no avail. I've been trying to use `api.tlm(...)` as shown in the button documentation, but I realize it returns a promise and I don't know how to get the actual value back into the widget.

ryanmelt, 2023-10-03:
Yes... we have at least most of the same canvas widgets that react to telemetry implemented.  

Like CANVASLINEVALUEWIDGET:
https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-init/plugins/packages/openc3-tool-common/src/components/widgets/CanvaslinevalueWidget.vue

tsjordan-eng, 2023-10-03:
I don't understand how to set the position equal to telemetry values from the docs, just the color.

https://docs.openc3.com/docs/v5/telemetry-screens#canvaslinevalue

jmthomas, 2023-10-17:
You can't change the position ... you hide and show lines by changing the color to match the background.

closed 2023-10-17 by jmthomas