OpenC3/cosmos #852: Inject raw bytes into an interface for reading

davimnz, 2023-10-02:
Dear OpenC3 team,

What is the appropriate way to inject raw bytes into an interface for reading? I'm aware of the `ìnject_tlm` function, but I don't have the target's name nor the packet's name when reading the bytes. Is there a function similar to `send_raw` of `cmd_api` but for telemetries? I know we can achieve this by reading raw bytes from an outside source (e.g., Redis) and returning them in a protocol, but I was wondering if this is the proper way to do it.

Thanks

ryanmelt, 2023-10-02:
There is no generic way to inject raw bytes into the interface. 

Interfaces block on the socket, or serial port, or whatever, and the only way to wake them up is to get real data on that device.  So if you try to do something with a protocol... it will generally still be blocked on the device until real data comes in.

One option is to create another COSMOS interface that can act as the other side of the connection.  So create a TCP/IP server that can inject data back to your TCP/IP client (for example).   Then to use it you would connect your interface to the TCP/IP Server rather than the real device and send a command out the TCP/IP server interface that would end up as raw bytes at the TCP/IP client.

closed 2023-10-02 by ryanmelt

ryanmelt, 2023-10-09:
Generally that would use two Intefaces.  One for the satellite connection, and another to read the file.
You can map the same target to both interfaces.  The file target would be just a MAP_TLM_TARGET.
https://docs.openc3.com/docs/v5/plugins#map_tlm_target