OpenC3/cosmos #865: Python import path

apogeecmb, 2023-10-12:
I was reviewing the updated Script Writing Guide.  I definitely appreciate the continuing work you all are doing to update those parts of the documentation to reflect the new Python scripting capability.  It mentions using standard Python import statements to include other libraries.  I'm curious what the proper path is for imports.  For example, if I'm trying to import a file that's in my target's lib directory, I tried `import [TARGET_NAME].lib.[filename]`, but I couldn't get that to work for me.  What's the correct way to do an import from a target's lib or procedures directory?  Thanks.

ryanmelt, 2023-10-18:
Still working on how to best support this.   Currently we only have load_utility implemented for python, but we need this before declaring Python done.

closed 2023-10-22 by jmthomas

apogeecmb, 2023-10-22:
So I gather this has been implemented now but how exactly?  What is the PYTHONPATH that libraries can be imported from?  The target directory or the host system PYTHONPATH?

jmthomas, 2023-10-23:
We hacked into the python import system so target directory paths can be imported. It works just like regular imports:

```
import TARGET.lib.file
# or
from TARGET.lib.file import Class
```

Since COSMOS target names must be capitalized this allow us to distinguish between import paths and our own.

PR #886 (2023-10-22): Import target files in Python
closes #865