OpenC3/cosmos #873: ./openc3.sh build stuck on 122/166 with caniuse-lite is outdated

haisamido, 2023-10-16:
I ran:

`./openc3.sh build`

## Which yields the below and is stuck on 122/166
```bash
[+] Building 31.1s (121/166)                                                                                                                                                                                                                                         
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 11/27] COPY ./plugins/packages/openc3-cosmos-tool-dataextractor/*.json packages/openc3-cosmos-tool-dataextractor/                                                                                            0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 12/27] COPY ./plugins/packages/openc3-cosmos-tool-dataviewer/*.json packages/openc3-cosmos-tool-dataviewer/                                                                                                  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 13/27] COPY ./plugins/packages/openc3-cosmos-tool-handbooks/*.json packages/openc3-cosmos-tool-handbooks/                                                                                                    0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 14/27] COPY ./plugins/packages/openc3-cosmos-tool-limitsmonitor/*.json packages/openc3-cosmos-tool-limitsmonitor/                                                                                            0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 15/27] COPY ./plugins/packages/openc3-cosmos-tool-packetviewer/*.json packages/openc3-cosmos-tool-packetviewer/                                                                                              0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 16/27] COPY ./plugins/packages/openc3-cosmos-tool-scriptrunner/*.json packages/openc3-cosmos-tool-scriptrunner/                                                                                              0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 17/27] COPY ./plugins/packages/openc3-cosmos-tool-tablemanager/*.json packages/openc3-cosmos-tool-tablemanager/                                                                                              0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 18/27] COPY ./plugins/packages/openc3-cosmos-tool-tlmgrapher/*.json packages/openc3-cosmos-tool-tlmgrapher/                                                                                                  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 19/27] COPY ./plugins/packages/openc3-cosmos-tool-tlmviewer/*.json packages/openc3-cosmos-tool-tlmviewer/                                                                                                    0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 20/27] COPY ./plugins/packages/openc3-tool-common/ packages/openc3-tool-common/                                                                                                                              0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 21/27] COPY ./plugins/packages/openc3-cosmos-ace-diff/ packages/openc3-cosmos-ace-diff/                                                                                                                      0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 22/27] COPY ./plugins/packages/openc3-cosmos-demo/*.json packages/openc3-cosmos-demo/                                                                                                                        0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 23/27] RUN yarn config set registry https://registry.npmjs.org && yarn --network-timeout 600000                                                                                                              0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 24/27] COPY ./plugins/docker-package-build.sh ./plugins/docker-package-install.sh ./plugins/babel.config.js ./plugins/.eslintrc.js ./plugins/.nycrc ./                                                       0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 25/27] RUN chmod +x ./docker-package-build.sh ./docker-package-install.sh                                                                                                                                    0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 26/27] COPY ./plugins/openc3-tool-base/ openc3-tool-base/                                                                                                                                                    0.0s
 => [openc3-cosmos-init openc3-frontend-tmp 27/27] RUN ["/openc3/plugins/docker-package-install.sh", "openc3-tool-base"]                                                                                                                                       29.5s
 => => # --- packageInstall openc3-tool-base yarn build                                                                                                                                                                                                             
 => => # yarn run v1.22.19                                                                                                                                                                                                                                          
 => => # $ webpack --mode=production                                                                                                                                                                                                                                
 => => # Browserslist: caniuse-lite is outdated. Please run:                                                                                                                                                                                                        
 => => #   npx update-browserslist-db@latest                                                                                                                                                                                                                        
 => => #   Why you should do it regularly: https://github.com/browserslist/update-db#readme                                                                                                                                                                         
```

ryanmelt, 2023-10-16:
Doesn't appear stuck...  The build takes a while, much longer than 29.5 seconds.
If it is stuck, then you probably need to give more resources to Docker Desktop.

haisamido, 2023-10-16:
> Doesn't appear stuck... The build takes a while, much longer than 29.5 seconds. If it is stuck, then you probably need to give more resources to Docker Desktop.

That 31.1 s is after I reran it. Previously it was hundreds of seconds. I thin I have sufficient docker resources

[image]

ryanmelt, 2023-10-16:
You have plenty of resources.  The build takes 10-15 minutes.

ryanmelt, 2023-10-16:
Also, you don't need to build unless you are changing something, or developing the core code.
Just run: ./openc3.sh run

closed 2023-10-17 by ryanmelt

haisamido, 2023-10-17:
@ryanmelt it's been over 49000 seconds and still not running

[+] Building 49160.0s (121/166)                                                                                                     
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 20/27] COPY ./plugins/packages/openc3-tool-common/ packages/openc3-tool-co  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 21/27] COPY ./plugins/packages/openc3-cosmos-ace-diff/ packages/openc3-cos  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 22/27] COPY ./plugins/packages/openc3-cosmos-demo/*.json packages/openc3-c  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 23/27] RUN yarn config set registry https://registry.npmjs.org && yarn --n  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 24/27] COPY ./plugins/docker-package-build.sh ./plugins/docker-package-ins  0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 25/27] RUN chmod +x ./docker-package-build.sh ./docker-package-install.sh   0.0s
 => CACHED [openc3-cosmos-init openc3-frontend-tmp 26/27] COPY ./plugins/openc3-tool-base/ openc3-tool-base/                   0.0s
 => [openc3-cosmos-init openc3-frontend-tmp 27/27] RUN ["/openc3/plugins/docker-package-install.sh", "openc3-tool-base"]   49157.5s
 => => # --- packageInstall openc3-tool-base yarn build                                                                            
 => => # yarn run v1.22.19                                                                                                         
 => => # $ webpack --mode=production                                                                                               
 => => # Browserslist: caniuse-lite is outdated. Please run:                                                                       
 => => #   npx update-browserslist-db@latest                                                                                       
 => => #   Why you should do it regularly: https://github.com/browserslist/update-db#readme

ryanmelt, 2023-10-17:
Yeah, that's not right...
We do lots of builds everyday, so its something unique on your end.
Something might be messed up with Docker so you could try "docker system prune --all" (will delete all your local containers though).