OpenC3/cosmos #875: Fragmented Fields

timothyscherer, 2023-10-17:
I have an application where a stream of bytes gets converted to dwords, where each dword contains a single byte. The bytestream contains data that is packed by bits, so in certain cases I may need fields to span multiple dwords, but only care about the single byte they contain. Is there a method for dealing with this already built in?

jmthomas, 2023-10-17:
You can absolutely define items that overlap other items. COSMOS will print a warning if you do unless you add the [OVERLAP](https://docs.openc3.com/docs/v5/telemetry#overlap) keyword to the item definition.

ryanmelt, 2023-10-17:
You can also specify "gaps" and ignore middle data, though that usually isn't recommended.  The recommendation is to just create "RESERVED1" or UNUSED1 type item names for the filler data.   You can then IGNORE these reserved items in your target.txt configuration.

timothyscherer, 2023-10-17:
So say I have these 2 bytes: 0xabcd.
In the packet I receive, that cargo looks like 0x000000ab000000cd.

As an example, the field I need to parse spans bit 15 down to bit 4, which encompasses 0xabc.
How do I define a single telemetry item that collects 0xabc as a 12-bit UINT?

jmthomas, 2023-10-17:
Sounds like you need to create a [derived item](https://docs.openc3.com/docs/v5/telemetry#derived-items) which combines two of your existing fields. To combine them into a simple UINT16 (0xabcd):

APPEND_ITEM PART1 32 UINT
APPEND_ITEM PART2 32 UINT
...
ITEM COMBINED 0 0 DERIVED
  GENERIC_READ_CONVERSION_START UINT 16
    packet.read('part1') << 8 | packet.read('part2')
  GENERIC_READ_CONVERSION_END

timothyscherer, 2023-10-17:
Perfect, thank you!

closed 2023-10-17 by timothyscherer