OpenC3/cosmos #879: Script Runner Improvements

tsjordan-eng, 2023-10-18:
The Script Runner has some great features, but there are some improvements that could make it more useful and easier to use.

* Script tabs and/or window splits: 
  * The tabs in COSMOS4 were nice. It's a little cumbersome to use browser tabs especially when using the "Execute Selection" right-click option. 
  * If this is too much to do for now, maybe add the script name to the browser tab title?
  * The little Filename combo-box is always grayed out for me. I don't know what it's supposed to do.
* Double-click to open: Right now we have to browse, click, then click the open button. Double-click is more intuitive.
* Recent files/directories: Similar to operating system file browsers, it might be nice to have some shortcuts to common places rather than having to drill down each time you want to open a file.
* Specify file in URL: This is nice for browser shortcuts. I see that other tools support this and it's super nice to use the browser's history and shortcuts to get to favorite pages.
* PyTest Support? More of a curiosity for now, but I'm wondering if this will be supported in the Suite Runner

tsjordan-eng, 2023-11-02:
Looks like the URL one is also tracked here: #905 👍

jmthomas, 2023-11-08:
Let me address some of your comments:

We don't use tabs in Script Runner basically because of browser tabs already. I added the script name to the tab title in #916 as well as put the filename in the url when opening a new file: `http://localhost:2900/tools/scriptrunner?file=INST/procedure/collect.rb`

The filename combo-box is always greyed out when editing. This is just showing you the name of the script you're working on. However when you run it changes to represent the currently loaded script. This is interesting as you `load_utility` and dive in and out of lower level scripts. If you pause you can change the drop down to view where the script is waiting in the higher level scripts.

I could get behind Double-click to open but it looks like it will have to wait for Vuetify 3 (#75).

Recent files is interesting. I could see having a File->"Open Recent >" which would open a menu to the right containing the latest 10 files or something like that.

We always looking to add better documentation and our Ace browser would probably support something like right click.

I'm not sure what PyTest support would look like in Script Runner. Our existing Test Suite framework is what we're using for building test suites.

PR #951 (2023-12-01): Add Open Recent to ScriptRunner, fix file icons
closes #879
closed 2023-12-01 by jmthomas