OpenC3/cosmos #882: Improve Command Description by using .json

JiveyGuy, 2023-10-18:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
The current format of commands looks unclear and could be formatted incorrectly. 

**Describe the solution you'd like**
Instead of using the current .txt format use .json or TOML. This way the format is clear. 

**Additional context**
Current format: 

```
COMMAND BBB EXAMPLE BIG_ENDIAN "Packet description"
  # Keyword           Name  BitSize Type   Min Max  Default  Description
  APPEND_ID_PARAMETER ID    16      INT    1   1    1        "Identifier"
  APPEND_PARAMETER    VALUE 32      FLOAT  0   10.5 2.5      "Value"
  APPEND_PARAMETER    BOOL  8       UINT   MIN MAX  0        "Boolean"
    STATE FALSE 0
    STATE TRUE 1
  APPEND_PARAMETER    LABEL 0       STRING          "OpenC3" "The label to apply"
```

jmthomas, 2023-10-18:
I think if we ever changed our format we would use YAML over JSON or TOML. This would require a great deal of work and it's just not on our road map at the moment.

jmthomas, 2023-10-18:
closing in favor of #883

closed 2023-10-18 by jmthomas