OpenC3/cosmos #884: Python Pip Libraries

tsjordan-eng, 2023-10-18:
How can we use other python libraries within the COSMOS script runner? Do we have to `pip install` in the container during build time or is there a way to drop in a requirements file that COSMOS takes care of at runtime?

ryanmelt, 2023-10-19:
This is available now if you run from the main branch.   Admin -> Gems has been replaced by Admin -> Packages where you can load Python packages.

ryanmelt, 2023-10-19:
Will be released in 5.12

closed 2023-10-19 by ryanmelt

apogeecmb, 2024-01-08:
Is this functionality described somewhere in the documentation?  I looked around and didn't see it, but maybe I was looking in the wrong place.  Do you just supply for example the whl file for any python library you want to have available?  And then OpenC3 will install it along with any other dependencies?  Thanks.

ryanmelt, 2024-01-10:
Yes exactly.  You can either install through the Admin->Packages section, or you can include a requirements.txt in your plugin and it will get installed when you install the plugin.
Documentation does still need to be updated.

tsjordan-eng, 2024-01-18:
When installing via the GUI, I get SSL errors. This can be caused by my company's proxy, but if we're uploading the `whl` files I wouldn't expect it to try to reach out to the internet; couldn't it just install from those files? (`numpy-1.19.5-cp36-cp36m-manylinux2010_x86_64.whl`)
```
Stdout:
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))) - skipping

Stderr:
WARNING: The directory '/openc3/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
ERROR: numpy-1.19.5-cp36-cp36m-manylinux2010_x86_64.whl is not a supported wheel on this platform.
```

Then I keep getting this "toast":
[image]

Of course I get the same error SSL with the requirements.txt method, but that's to be expected. However the plugin installation still says "Complete", not "WARNING". Clicking on the little eyeball my stderr logs are:
```
Stderr:
WARNING: The directory '/openc3/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))': /simple/orbit-predictor/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))': /simple/orbit-predictor/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))': /simple/orbit-predictor/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))': /simple/orbit-predictor/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))': /simple/orbit-predictor/
ERROR: Could not find a version that satisfies the requirement orbit_predictor (from versions: none)
ERROR: No matching distribution found for orbit_predictor
```

ryanmelt, 2024-01-18:
re #1065

tsjordan-eng, 2024-02-18:
I tried changing my "Pypi URL" in Admin Console to my company's mirror, but when I upload a .whl package in Packages in Admin Console it still tries the typical one. I did click "save" and refresheed the page to see it was there. Do I need to do anything else to get it to use the mirror?

Running Podman on RHEL 9.2, COSMOS 5.14.2. USER/GROUP are both `root`.

Error:
```
Stdout:
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)'))) - skipping

Stderr:
WARNING: The directory '/openc3/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
ERROR: numpy-1.19.5-cp36-cp36m-manylinux2010_x86_64.whl is not a supported wheel on this platform.
```