OpenC3/cosmos #889: Connecting to test equipment on different NIC

mattgermano, 2023-10-20:
I have a piece of test equipment connected to a secondary ethernet port on my system. The primary ethernet interface is `eno1` and the secondary ethernet interface is `ens4f3`.

The COSMOS compose configuration creates a bridge network by default without host networking. As a result, COSMOS can only connect to devices accessible from the primary ethernet interface (`eno1`) and not the test equipment connected to the secondary ethernet interface.

I attempted to `docker exec` into the operator container and confirmed that it is unable to ping the test equipment IP address on the secondary ethernet interface. 

Is there a recommended solution for this? I will have 10+ pieces of test equipment all connected via a secondary ethernet interface and will need to be able to connect to them from COSMOS.

Thank you.

ryanmelt, 2023-10-20:
I'll do some research on the Docker network aspect.   Alternatively you could also use a COSMOS bridge.  https://docs.openc3.com/docs/guides/bridges

ryanmelt, 2023-10-21:
This looks like one path:
https://medium.com/@havloujian.joachim/advanced-docker-networking-outgoing-ip-921fc3090b09

You also might be able to manually set a route on the host.

mattgermano, 2023-10-21:
Thanks Ryan. After some research, I found a relatively straightforward solution using the [`ipvlan`](https://docs.docker.com/network/drivers/ipvlan/) Docker networking driver. 

I created a new `ipvlan` network that binds to the secondary ethernet interface (`ens4f3`) using `docker network create`:
```
$ docker network create -d ipvlan \
    --subnet=192.168.1.0/24 \
    --gateway=192.168.1.1 \
     -o ipvlan_mode=l2 \
     -o parent=ens4f3 openc3-test-equip-network
```
The subnet and gateway were obtained from the network configuration on the secondary ethernet interface:
```
$ ip addr show ens4f3
3: ens4f3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    inet 192.168.1.250/24 brd 192.168.1.255 scope global ens4f3
```
I then connected the `openc3-openc3-operator-1` container to the new network using `docker network connect`:
```
$ docker network connect openc3-test-equip-network openc3-openc3-operator-1
```
Running `ip addr show` inside of the operator container now shows that it has a new ethernet interface with an IP address on the subnet specified in the `docker network create` command. This is in addition to the bridge network that the Docker compose configuration creates:
```
$ docker exec openc3-openc3-operator-1 ip addr show

57: eth0@if58: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue state UP
      inet 172.19.0.7/16 brd 172.19.255.255 scope global eth0
63: eth1@if6: <BROADCAST,MULTICAST,UP,LOWER_UP,M-DOWN> mtu 1500 qdisc noqueue state UNKNOWN
     inet 192.168.1.2/24 brd 192.168.1.255 scope global eth1
```
The operator container is now able to `ping` the test equipment IP address and COSMOS can successfully connect to it.

This network could also be created and added to the operator service in the `compose.yaml` file using a configuration like this:
```yml
networks:
  default:
    name: openc3-cosmos-network
  test-equip:
    name: openc3-test-equip-network
    driver: ipvlan
    driver_opts:
      ipvlan_mode: "l2"
      parent: "ens4f3"
    ipam:
      config:
        - subnet: 192.168.1.0/24
          gateway: 192.168.1.1

openc3-operator:
...
  networks:
      - default
      - test-equip
```

I will mark this issue as closed.

closed 2023-10-21 by mattgermano

ryanmelt, 2023-10-21:
Awesome and thanks for the great write up on your solution!