OpenC3/cosmos #896: Script Runner failing to Load

bradenvirdellgenevatech, 2023-10-25:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
When trying to load scripts in script runner we are getting this error `Error: Failed to connect to OpenC3. TypeError: Cannot read properties of undefined (reading 'push')` and it doesnt allow us to access any scripts

**To Reproduce**
1. Open script runner
2. Select Open File

**Expected behavior**
The File to open

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**

- OS: Ubuntu
- OpenC3 COSMOS Version [e.g. 5.9.0]
- Browser
  -- [*] Internet Explorer (IE)
  -- [*] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

ryanmelt, 2023-10-25:
Does anything print in the browser debug console?

bradenvirdellgenevatech, 2023-10-25:
Didn't check, we got it to stop doing it by removing and reuploading one of the our gems so I can't replicate again. If it pops up I will be sure to check the debug console next time

thomasstjeangenevatech, 2025-03-13:
Hello,

Re-engaging this thread because this issue just occurred again for us.

    OS: Ubuntu
    OpenC3 COSMOS 5.17.1
    Browser
    -- Edge
    -- Google Chrome
    -- [Mozilla Firefox

Same error message as previous, `TypeError: Cannot read properties of undefined (reading 'push')`. I am not seeing any errors in the browser console besides `Uncaught (in promise) Error: Access to storage is not allowed from this context.`, but this seems unrelated to the error as that message hasn't appeared any other time I have attempted to recreate.

[image]

Interestingly, on Firefox, I get a different message: `Error: Failed to connect to OpenC3. TypeError: e is undefined`, although still no message in the console. Additionally, this is not an issue for everyone on our team. Right now, 2 of us are unable to access scripts while everyone else can. Also, the 'Recently used' scripts are still accessible.

ryanmelt, 2025-03-13:
The Extension context invalidated errors are likely from some form of Chrome browser extension.  
Can you try disabling any browser extensions and see what changes?   That might also explain why it works for others.

ryanmelt, 2025-03-13:
Also the next thing I would want to see is the Network tab in the chrome Dev tools.    Are you receiving the replies from our backend and just the frontend is stuck?

thomasstjeangenevatech, 2025-03-13:
There is only one network request on the Network tab when I open the "File Open" dialog:
[image]

The message and status are the same as the request when it fetches the files successfully, but the size and time are significantly larger:
[image]

ryanmelt, 2025-03-13:
4.8MB is a lot!  I'm going to reclose this and refer to this ticket:
https://github.com/OpenC3/cosmos/issues/1950

We are planning on optimizing the file chooser to only pull the currently shown details rather than everything like it does now.
This is a priority so should be addressed soon.

closed 2025-03-13 by ryanmelt