OpenC3/cosmos #897: Ports not releasing when installing/uninstalling plugin

rmslick, 2023-10-25:
Currently have a plugin bind to port 5001 (forwarded in the yaml as 5001:5001). When trying to uninstall/reinstall plugin we are getting the following error. Any insight helpful

COMOTCP_INT: RuntimeError : Error binding to port 5001.
Either another application is using this port
or the operating system is being slow cleaning up.
Make sure all sockets/streams are closed in all applications,
wait 1 minute and try again.
/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.2/lib/openc3/interfaces/tcpip_server_interface.rb:323:in `rescue in start_listen_thread'
/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.2/lib/openc3/interfaces/tcpip_server_interface.rb:320:in `start_listen_thread'
/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.2/lib/openc3/interfaces/tcpip_server_interface.rb:129:in `connect'
interface_microservice.rb:609:in `connect'
interface_microservice.rb:457:in `block in run'
interface_microservice.rb:455:in `synchronize'
interface_microservice.rb:455:in `run'
/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.2/lib/openc3/microservices/microservice.rb:55:in `run'
interface_microservice.rb:684:in `<main>'

ryanmelt, 2023-10-25:
Our code properly handles reusing sockets.  This might be something with an old Docker version?

rmslick, 2023-10-26:
Okay, heres some more info in case you can provide insights. We have one machine on which cosmos is running cleanly without this issue. And two that have this issue. All 3 run rhel8 and use a proxy.

_Docker versions:_
Docker version on working machine is 23.0.6 and nonworking 24.0.6. 

_Blind troubleshooting_
On working machine, when we remove the plugin, the docker-proxy process on the host machine remains bound to port 5001, but the ruby process inside the openc3-operator1 docker container releases port 5001 inside that container. 

On non working machines, within the Docker container the ruby process does initially release the internal docker port 5001, but a new ruby process is immediately spawned that binds to that port (the process number on the port changes). 

Any guesses? Thanks in advance

ryanmelt, 2023-10-26:
Sounds like the nonworking ones are maybe crashing and then respawning and reconnecting?  But if you remove the plugin then they shouldn't be respawned at all.

Can you look at the docker logs for the openc3-operator container and see if their are any crash logs?

ryanmelt, 2023-10-26:
Also the next thing to do would be to update to our latest release 5.11.3 and see if that resolves your issue.  We've made a lot of updates since 5.5.2

closed 2023-11-01 by ryanmelt