OpenC3/cosmos #898: Gem Mod Date/Times Are Changing

SeanEsslinger, 2023-10-25:
While troubleshooting a gem issue, I've noticed a couple oddities:
1) The create/modification date/time changes every minute for all gems under the /gems/gems directory in the cmd-tlm-api container.  The following is two listing ("ls") commands performed six minutes apart.  What lead to looking for this was when down inside the openc3-cosmos-spw gem directory, the "ls" command would show files/directories and then a few seconds later the directory appeared to be empty.  A change directory back into the same gem directory (i.e., "cd .") resolves the issues for about one minute.  It acts as though the directory is being deleted and replaced with an exact copy of the original.  This behavior is also seen in the bit-struct-0.17 directory.
/gems/gems $ ls -al
total 92
drwxr-xr-x   23 305226   2260          4096 Oct 25 21:02 .
drwxrwxrwx   10 openc3   openc3        4096 Oct 18 23:15 ..
drwxr-xr-x    5 305226   2260          4096 Oct 25 21:02 bit-struct-0.17
drwxr-xr-x    5 305226   2260          4096 Oct 25 21:02 openc3-cosmos-demo-5.11.3
drwxr-xr-x    4 305226   2260          4096 Oct 25 21:02 openc3-cosmos-spw-1.0.12
drwxr-xr-x    5 305226   2260          4096 Oct 25 21:02 openc3-cosmos-spw-1.0.71
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-admin-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-autonomic-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-bucketexplorer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-calendar-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-cmdsender-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-cmdtlmserver-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-dataextractor-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-dataviewer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-handbooks-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-limitsmonitor-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-packetviewer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-scriptrunner-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-tablemanager-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-tlmgrapher-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-cosmos-tool-tlmviewer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:02 openc3-tool-base-5.11.3
drwxr-xr-x    6 305226   2260          4096 Oct 25 21:02 rake-compiler-1.2.5
/gems/gems $ ls -al
total 92
drwxr-xr-x   23 305226   2260          4096 Oct 25 21:08 .
drwxrwxrwx   10 openc3   openc3        4096 Oct 18 23:15 ..
drwxr-xr-x    5 305226   2260          4096 Oct 25 21:08 bit-struct-0.17
drwxr-xr-x    5 305226   2260          4096 Oct 25 21:08 openc3-cosmos-demo-5.11.3
drwxr-xr-x    4 305226   2260          4096 Oct 25 21:08 openc3-cosmos-spw-1.0.12
drwxr-xr-x    5 305226   2260          4096 Oct 25 21:08 openc3-cosmos-spw-1.0.71
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-admin-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-autonomic-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-bucketexplorer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-calendar-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-cmdsender-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-cmdtlmserver-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-dataextractor-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-dataviewer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-handbooks-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-limitsmonitor-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-packetviewer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-scriptrunner-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-tablemanager-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-tlmgrapher-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-cosmos-tool-tlmviewer-5.11.3
drwxr-xr-x    3 305226   2260          4096 Oct 25 21:08 openc3-tool-base-5.11.3
drwxr-xr-x    6 305226   2260          4096 Oct 25 21:08 rake-compiler-1.2.5

Is this expected behavior?  If so, why?

2) There's an old gem version that keeps coming back after being deleted.  Version 1.0.12 of the openc3-cosmos-spw gem is gone after using "gem cleanup" or "gem uninstall" to delete it, the directory is gone and "gem list --local" shows the gem is gone.  After about one minute, something seems to reinstall the gem (which appears to be related to the first observation).  Why does it get reinstalled and where is this version of the gem file being cached?

ryanmelt, 2023-10-25:
Is your init container spinning?  Do a "docker ps" and check the started time on the openc3-cosmos-init container.

SeanEsslinger, 2023-10-25:
LOL - I was just looking at the container and noticed it is spinning.  What would cause that?

ryanmelt, 2023-10-25:
docker logs <init container hash>
Generally it is some sort of permissions issue

SeanEsslinger, 2023-10-25:
Interesting.  It looks like maybe there was a load error back with 1.0.12 that has been stuck in there since that version.  From the Admin tab, I deleted the SPW gem; had to do it a couple times as it appears to have deleted the 1.0.71 and then 1.0.12 showed back up.  I now have no gems loaded and the init container has stopped restarting.

After waiting a few minutes to ensure the init container didn't start looping again, I reloaded the latest SPW gem and appears to be working as expected.  

Thanks for the quick response!

Docker Log:
2023-10-25T21:34:31Z waiting for Minio http://openc3-minio:9000 RC: 0
2023-10-25T21:34:32Z waiting for Redis. RC: 0
2023-10-25T21:34:33Z waiting for Redis Ephemeral. RC: 0
Checking for needed migrations...
Migrations complete
Added `openc3minio` successfully.
Created policy `script` successfully.
Added user `scriptrunnerminio` successfully.
mc: <ERROR> Unable to make user/group policy association. The specified policy change is already in effect. (Specified policy update has no net effect).
No version change detected for: openc3-tool-base-5.11.3.gem__20231013163532
No version change detected for: openc3-cosmos-tool-admin-5.11.3.gem__20231013163533
Local init running: /plugins exists
Local init found scope: DEFAULT
Local init found plugin_dir: /plugins/DEFAULT/openc3-cosmos-spw-1
Error installing plugin: DEFAULT: /plugins/DEFAULT/openc3-cosmos-spw/openc3-cosmos-spw-1.0.12.gem

RuntimeError : DEFAULT__openc3_targets:SPW already exists at create
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/model.rb:141:in `create'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/plugin_model.rb:214:in `block in install_phase2'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/config/config_parser.rb:217:in `parse_loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/config/config_parser.rb:217:in `parse_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/plugin_model.rb:208:in `install_phase2'
/openc3/bin/openc3cli:466:in `load_plugin'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:75:in `block (2 levels) in local_init'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:59:in `each_child'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:59:in `block in local_init'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:56:in `each_child'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:56:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:56:in `local_init'
/openc3/bin/openc3cli:749:in `<main>'

plugin.txt20231025-106-este37:9: MAP_TARGET SPW
Error: Unknown keyword and parameters for Target: MAP_TARGET SPW
Backtrace:
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/target_model.rb:548:in `handle_config'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/plugin_model.rb:222:in `block in install_phase2'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/config/config_parser.rb:217:in `parse_loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/config/config_parser.rb:217:in `parse_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/plugin_model.rb:208:in `install_phase2'
/openc3/bin/openc3cli:466:in `load_plugin'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:75:in `block (2 levels) in local_init'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:59:in `each_child'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:59:in `block in local_init'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:56:in `each_child'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:56:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/utilities/local_mode.rb:56:in `local_init'
/openc3/bin/openc3cli:749:in `<main>'
Updating existing plugin: openc3-cosmos-spw-1.0.71.gem__20231025200839 with openc3-cosmos-spw-1.0.71.gem
No changes detected - Exiting without change
Local init found plugin_dir: /plugins/DEFAULT/openc3-cosmos-spw
Loading new plugin: /plugins/DEFAULT/openc3-cosmos-spw/openc3-cosmos-spw-1.0.12.gem
{"name"=>"openc3-cosmos-spw-1.0.12.gem__20231013222825", "variables"=>{"spw_target_name"=>"SPW", "spw_target_ip"=>"host.docker.internal"}, "plugin_txt_lines"=>["# Set VARIABLEs here to allow variation in your plugin", "", "VARIABLE spw_target_name SPW", "#VARIABLE spw_target_ip 10.10.1.50 # Orig: host.docker.internal", "VARIABLE spw_target_ip host.docker.internal", "", "TARGET SPW <%= spw_target_name %>", "INTERFACE <%= spw_target_name %>_INT spw_interface.rb host.docker.internal 8010 8011 10.0 nil BURST", "  MAP_TARGET <%= spw_target_name %>"], "needs_dependencies"=>true, "updated_at"=>1697236105768921419}
Wed Oct 25 21:35:38 UTC 2023
Restoring gems to pristine condition...
Restored bit-struct-0.17
Restoring gems to pristine condition...

closed 2023-10-26 by ryanmelt