OpenC3/cosmos #907: Serial Port Permissions Podman

tsjordan-eng, 2023-11-02:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Interface can't connect to serial port even when I can read/write manually in container

**To Reproduce**
Steps to reproduce the behavior:

Make serial interface and try to connect

plugin.txt
```
...
INTERFACE <%= uut_target_name %>_SER serial_interface.rb <%= serial_port %> <%= serial_port %> <%= baud_rate %> NONE 1 10.0 nil LENGTH 48 32 10 1 BIG_ENDIAN 0 0x1ACFFC1D
  MAP_TLM_TARGET <%= uut_target_name %>
```

Using Podman. Add SELinux exception: `sudo setsebool -P container_use_devices 1`

Compose, Add devices:
```
openc3-operator:
...
  group_add:
    - '18' # dialout
    - 'dialout'
  devices:
    - "/dev/ttyUSB0"
...
```

**Expected behavior**

It should read/write data from the serial port.

I can manually read/write data with `cat /dev/ttyUSB0` and `echo 'hello' > /dev/ttyUSB0` running *inside* the operator container with user 1000.

**Screenshots**

Error Message:
```
UUTDEV_SER: EACCES : Permission denied @ rb_sysopen - /dev/ttyUSB0
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/io/posix_serial_driver.rb:58:in `initialize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/io/posix_serial_driver.rb:58:in `open'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/io/posix_serial_driver.rb:58:in `initialize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/io/serial_driver.rb:69:in `new'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/io/serial_driver.rb:69:in `initialize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/streams/serial_stream.rb:80:in `new'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/streams/serial_stream.rb:80:in `initialize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/interfaces/serial_interface.rb:70:in `new'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/interfaces/serial_interface.rb:70:in `connect'
interface_microservice.rb:616:in `connect'
interface_microservice.rb:464:in `block in run'
interface_microservice.rb:462:in `synchronize'
interface_microservice.rb:462:in `run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/microservices/microservice.rb:55:in `run'
interface_microservice.rb:691:in `<main>'
```

**Environment (please complete the following information):**

- OS: RHEL9
- Container: Podman
- OpenC3 COSMOS Version 5.11.3
- Browser
  -- [] Internet Explorer (IE)
  -- [] Google Chrome
  -- [x] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

tsjordan-eng, 2023-11-02:
welp, I re-read https://docs.openc3.com/docs/guides/bridges and it says to give permissions to all users for the device on the host:
```bash
sudo chmod 666 /dev/ttyUSB0
```
That worked...

But I don't know why I was able to read/write to it within the container with `podman exec ...`, but the Interface service couldn't. Is _this_ a bug?

ryanmelt, 2023-11-02:
The chmod is just a quick fix.  Permissions are fun.

On rootless podman we generally run our containers as root (which turns into the userid of the user on the host that is running).

If you ran the container by itself as user 1000 that would be different than our standard rootless way of running.

tsjordan-eng, 2023-11-02:
Yeah, I agree that it's a quick fix. I *am* able to cat from and echo to the serial port _inside_ the operator container _as_ the user 1000. However it just seems that the interface service doesn't have permissions. Does that help?

In my `compose.yaml` my container has `user: 0:0` so I think I'm actually running it as my user which has the right permissions (part of the `dialout` group).

ryanmelt, 2023-11-03:
When you exec into the container and look at the dev, what owner / permissions do you see?

ls -la /dev/ttyUSB0

tsjordan-eng, 2023-11-03:
I am, in fact, running my containers as user `0`, but the read/write still works with uid `1000`.

```bash
~/lib/openc3/operators # ls -la /dev/ttyUSB3
crw-rw----    1 nobody   nobody    188,   3 Nov  3 16:21 /dev/ttyUSB3
```

As far as the `nobody` users, I see that artifact [here ](https://www.redhat.com/sysadmin/files-devices-podman )in the addendum. (I tried running an independent container the way they indicate, but get the same artifact). That article also makes me think my `group_add` line in the compose file I showed above is meaningless because it is translated inside the container and no longer corresponds to the host group. I haven't found a way to use `keep-groups` in a compose file.

That said, using `docker-compose exec -it openc3-operator cat /dev/ttyUSB3` receives bytes and `docker-compose exec -it openc3-operator echo "hello" > /dev/ttyUSB3` sends bytes even with the permissions shown in this comment. I don't know why the service inside the container can't do that without the `o+rw` permissions.

ryanmelt, 2023-11-04:
If you do ls with -lan you can see the numeric userid/groupid.   
0 should work, but I wouldn't expect 1000 to unless you did the chmod

ryanmelt, 2023-11-08:
Closing as the chmod worked, and I don't have a good environment to further debug.  Could be something with SELinux.

closed 2023-11-08 by ryanmelt