OpenC3/cosmos #908: Interface can't find protocol when target gets unmapped

tsjordan-eng, 2023-11-02:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**

I have a plugin with one target and two interfaces, one serial and the other TCP. Both use the same custom protocol. Upon installation I have the target mapped to the ethernet interface and tlm-only mapped to the serial interface. This works fine.

However, when I use `map_target_to_interface` to switch it completely to one interface or the other I begin seeing errors that my custom protocol file cannot be found for the unmapped interface and the unmapped interface also disappears from the Interfaces tab in the Command and Telemetry Server. I can stop the error messages if I map the target's tlm to the old interface so that it's attached to both of them again.

**To Reproduce**

1. Create a plugin with one target and two interfaces which use a custom protocol in `./TARGET/lib/<protocol>.rb`
3. Map the target to just one interface
4. See errors

**Expected behavior**

I should be able to freely map the target to whichever interface I want to use and get no errors. Also both interfaces should remain in the Interfaces tab of the Command and Telemetry Server.

**Screenshots**

Error messages:

```
2023-11-02 15:44:48.904	
ERROR	DEFAULT__INTERFACE__UUT2DEV_SER	
Unable to require uut_crc_protocol.rb due to cannot load such file -- uut_crc_protocol.rb. Ensure uut_crc_protocol.rb is in the OpenC3 lib directory.
```

```
2023-11-02 15:44:50.433	
ERROR	MicroserviceOperator	
Unexpected process died... respawning! ruby interface_microservice.rb DEFAULT__INTERFACE__UUT2DEV_SER
Stdout:
{"time":1698961486403794989,"@timestamp":"2023-11-02T21:44:46.403+00:00","severity":"INFO","container_name":"c22e5aeafc36","log":"Microservice running from: ruby plugin_microservice.rb ","type":"log"}
{"time":1698961486427883901,"@timestamp":"2023-11-02T21:44:46.427+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"Microservice initialized with config:
{\"name\"=>\"DEFAULT__INTERFACE__UUT2DEV_SER\", \"folder_name\"=>nil, \"cmd\"=>[\"ruby\", \"interface_microservice.rb\", \"DEFAULT__INTERFACE__UUT2DEV_SER\"], \"work_dir\"=>\"/openc3/lib/openc3/microservices\", \"ports\"=>[], \"env\"=>{}, \"topics\"=>[], \"target_names\"=>[], \"options\"=>[], \"parent\"=>nil, \"container\"=>nil, \"updated_at\"=>1698961344917133459, \"plugin\"=>\"openc3-cosmos-uut-0.2.0M.20231102145351.gem__20231102205452\", \"needs_dependencies\"=>false, \"secrets\"=>[], \"prefix\"=>nil}","type":"log"}
{"time":1698961487124725532,"@timestamp":"2023-11-02T21:44:47.124+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"Ruby microservice DEFAULT__INTERFACE__UUT2DEV_SER file interface_microservice.rb passed syntax check
","type":"log"}
{"time":1698961488211972744,"@timestamp":"2023-11-02T21:44:48.211+00:00","severity":"INFO","container_name":"c22e5aeafc36","log":"Microservice running from: ruby interface_microservice.rb DEFAULT__INTERFACE__UUT2DEV_SER","type":"log"}
{"time":1698961488226880487,"@timestamp":"2023-11-02T21:44:48.226+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"Microservice initialized with config:
{\"name\"=>\"DEFAULT__INTERFACE__UUT2DEV_SER\", \"folder_name\"=>nil, \"cmd\"=>[\"ruby\", \"interface_microservice.rb\", \"DEFAULT__INTERFACE__UUT2DEV_SER\"], \"work_dir\"=>\"/openc3/lib/openc3/microservices\", \"ports\"=>[], \"env\"=>{}, \"topics\"=>[], \"target_names\"=>[], \"options\"=>[], \"parent\"=>nil, \"container\"=>nil, \"updated_at\"=>1698961344917133459, \"plugin\"=>\"openc3-cosmos-uut-0.2.0M.20231102145351.gem__20231102205452\", \"needs_dependencies\"=>false, \"secrets\"=>[], \"prefix\"=>nil}","type":"log"}
{"time":1698961488904293629,"@timestamp":"2023-11-02T21:44:48.904+00:00","severity":"ERROR","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"Unable to require uut_crc_protocol.rb due to cannot load such file -- uut_crc_protocol.rb. Ensure uut_crc_protocol.rb is in the OpenC3 lib directory.","type":"log"}
{"time":1698961488904801710,"@timestamp":"2023-11-02T21:44:48.904+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"DEFAULT__INTERFACE__UUT2DEV_SER: shutdown requested","type":"log"}
{"time":1698961488904980995,"@timestamp":"2023-11-02T21:44:48.904+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"DEFAULT__INTERFACE__UUT2DEV_SER: stop requested","type":"log"}
{"time":1698961488905117818,"@timestamp":"2023-11-02T21:44:48.905+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"Shutting down microservice: DEFAULT__INTERFACE__UUT2DEV_SER","type":"log"}
{"time":1698961488916103544,"@timestamp":"2023-11-02T21:44:48.916+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__UUT2DEV_SER","container_name":"c22e5aeafc36","log":"Shutting down microservice complete: DEFAULT__INTERFACE__UUT2DEV_SER","type":"log"}

Stderr:
<internal:/usr/local/lib/site_ruby/3.2.0/rubygems/core_ext/kernel_require.rb>:85:in `require': Unable to require uut_crc_protocol.rb due to cannot load such file -- uut_crc_protocol.rb. Ensure uut_crc_protocol.rb is in the OpenC3 lib directory. (LoadError)
	from <internal:/usr/local/lib/site_ruby/3.2.0/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/top_level.rb:557:in `require'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/top_level.rb:442:in `require_file'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/top_level.rb:430:in `require_class'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/interface_model.rb:185:in `block in build'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/interface_model.rb:184:in `each'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/interface_model.rb:184:in `build'
	from interface_microservice.rb:361:in `initialize'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/microservices/microservice.rb:51:in `new'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/microservices/microservice.rb:51:in `run'
	from interface_microservice.rb:691:in `<main>'
<internal:/usr/local/lib/site_ruby/3.2.0/rubygems/core_ext/kernel_require.rb>:85:in `require': cannot load such file -- uut_crc_protocol.rb (LoadError)
	from <internal:/usr/local/lib/site_ruby/3.2.0/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/top_level.rb:557:in `require'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/top_level.rb:442:in `require_file'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/top_level.rb:430:in `require_class'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/interface_model.rb:185:in `block in build'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/interface_model.rb:184:in `each'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/models/interface_model.rb:184:in `build'
	from interface_microservice.rb:361:in `initialize'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/microservices/microservice.rb:51:in `new'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.3/lib/openc3/microservices/microservice.rb:51:in `run'
	from interface_microservice.rb:691:in `<main>'
```

**Environment (please complete the following information):**

- OS: RHEL 9 
- OpenC3 COSMOS Version 5.11.3
- Browser: Edge
- Container: Podman

ryanmelt, 2023-11-09:
If you completely unmap a target from an interface, then it will remove the targets files and this will happen.

Options:
1. Move the protocol to a top level lib folder in the plugin - It will then be available to all interfaces
2. Or leave the target mapped just as a telemetry target, and disconnect the interface.  In general the key is moving the commanding, not telemetry.

closed 2023-11-09 by ryanmelt

tsjordan-eng, 2023-12-21:
Both your options work. Thanks!