OpenC3/cosmos #909: Telemetry Configuration Dropdown

timothyscherer, 2023-11-03:
I have found it somewhat inconvenient to switch between configurations. It would be nice if the "Telemetry Viewer" sidebar link had a dropdown similar to the "Utilities", which contains the available configurations. So you can still use the button as it works now, but if you want a specific config you can just drop down the arrow and open other configs in new tabs or the current tab.

[image]

jmthomas, 2023-11-03:
Once you save and load a config you can see it in the url: `http://localhost:2900/tools/tlmviewer/?config=test`. In addition a local cookie is saved so the next time you go back to the tool it will automatically load the last config.

timothyscherer, 2023-11-03:
Yes I was just offering something a little more user friendly from my perspective

jmthomas, 2023-11-03:
I don't think we'll add this for everyone. If you want you can add a specific link to the tool menu by going to the Admin->Tools (http://localhost:2900/tools/admin/tools) tab and create a link to your specific configuration.

closed 2023-11-03 by jmthomas