OpenC3/cosmos #910: Plugin Workflow

rmslick, 2023-11-03:
I am trying to understand the recommended workflow or best practices for plugin development and finding it a little awkward. 

First - Following from your getting started guide, I run `openc3.bat cli generate plugin BOB` followed by `cd openc3-cosmos-bob` and `openc3.bat cli generate target BOB` The result from these two is: `.\openc3.bat : The term '.\openc3.bat' is not recognized as the name of a cmdlet, function, script file, or operable program.` Which makes sense because there is no such shell script in bobs plugin. Is there supposed to be?

Second - Is the recommended workflow for plugin development to:

      1. Generate plugin inside cosmos project
      2. Move plugin out of project source, into its own repo and develop there
      3. Upload either with admin console or load utility in openc3 batch script, assuming this scriptis present in plugin root

Or is it to generate plugin, develop and load from the openc3 project and use the project source files as the sole repo to track

ryanmelt, 2023-11-03:
You can develop your plugins in whatever folder you want.  The step you are missing is making sure the path the has openc3.sh or openc3.bat is in your computer's PATH environment variable.

ryanmelt, 2023-11-03:
We'll add that step to the Getting Started documentation

rmslick, 2023-11-03:
Great, thanks. Maybe this is also an env variable issue. When trying to run `openc3.bat  cliroot load bob-1.1.0.gemliroot load bob-1.1.0.gem` to load the plugin into the environment from the command line I get the following error: `Unable to resolv api hostname: openc3-cosmos-cmd-tlm-api`. Note that in this scenario, I am working with a plugin that is outside of the project folder.

ryanmelt, 2023-11-03:
That should work, you should need to use cliroot though, just use cli.

The key to that working is the network being added into the docker run line.   Our 5.11.3 openc3.sh script does that though.

rmslick, 2023-11-08:
Okay, now load is working. Is there a command line option to remove plugins by gemname?

ryanmelt, 2023-11-09:
Yes.  You can use openc3.bat cli unload full_plugin_name_with_instance_timestamp
See the plugins tab for the full name with __timestamp.
There is no way to query names yet with the cli tool, that is a future enhancement.

I'll leave this ticket open, we can enhance this with a flag that has it remove all instances without having to specify the timestamp.

jmthomas, 2024-04-16:
You can now list the plugins once https://github.com/OpenC3/cosmos/pull/1204 is closed. This will make it easy to get the full filename for closing.

closed 2024-04-16 by jmthomas