OpenC3/cosmos #911: Adding tool encounters HTTP 500

timothyscherer, 2023-11-06:
From the admin console, I am trying to add tools (under the tools tab) using the fields at the top of the page.

I first added a tool named "test" with the url "www.google.com", and it worked correctly.

[image]

Then, tried to create a tool named "test2" with the same configuration otherwise, and it fails.

[image]

I also tried renaming the "Telemetry Viewer" tool, which also causes issues, but I have no issue changing the link to the tool, or putting it in a category.

[image]

I am using this as suggested in #909, as I think this is a wonderful feature!

- OS: Ubuntu 22.04
- OpenC3 COSMOS Version 5.11.3
- Browser
  - [ ] Internet Explorer (IE)
  - [ ] Google Chrome
  - [ ] Chromium based browsers
  - [x] Mozilla Firefox
  - [ ] Safari

ryanmelt, 2023-11-06:
This is already fixed and will be in our next release:
Fixed as part of https://github.com/OpenC3/cosmos/pull/901

Older versions are effectively limited to one standalone link.

ryanmelt, 2023-11-06:
closed by #901

closed 2023-11-06 by ryanmelt