OpenC3/cosmos #919: XML <Code>AccessDenied</Code> when using ssh tunnel (openc3-minio issue?)

haisamido, 2023-11-09:
I successfully started Cosmos, via ./openc3.sh on my mac book laptop, and I am able to view simulated telemetry via [http://localhost:2900/](http://localhost:2900/) . Additionally I was also able to start Cosmos on an AWS EC2 RHEL8 instance. So I tunneled into the instance via `ssh -L 12900:127.0.0.1:2900 -N <MY EC2 IP ADDRESS>` (a method successfully used with other web portal applications) and tried this [http://localhost:12900/](http://localhost:12900/) and I was expecting to be able to view the telemetry from the remote EC2 instance, but I got the below:

```xml
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied.</Message>
<Key>base/index.html</Key>
<BucketName>tools</BucketName>
<Resource>/tools/base/index.html</Resource>
<RequestId>1795FED301D31540</RequestId>
<HostId>dd9025bab4ad464b049177c95eb6ebf374d3b3fd1af9251148b658df7ac2e3e8</HostId>
</Error>
```

The logs, on the remote instance, via the below command:
```bash
docker-compose logs -f \
  openc3-traefik \
  openc3-cosmos-cmd-tlm-api \
  openc3-cosmos-script-runner-api \
  openc3-operator \
  openc3-redis-ephemeral \
  openc3-minio openc3-redis \
  openc3-cosmos-init

```

```bash
cosmos-openc3-cosmos-cmd-tlm-api-1        | I, [2023-11-09T15:59:28.003089 #19]  INFO -- : [d41c4a39-0c86-4628-b0ee-72fdbc640a87] Started GET "/openc3-api/traefik" for 172.21.0.8 at 2023-11-09 15:59:28 +0000
cosmos-openc3-cosmos-cmd-tlm-api-1        | I, [2023-11-09T15:59:28.003792 #19]  INFO -- : [d41c4a39-0c86-4628-b0ee-72fdbc640a87] Processing by MicroservicesController#traefik as HTML
cosmos-openc3-cosmos-cmd-tlm-api-1        | I, [2023-11-09T15:59:28.004496 #19]  INFO -- : [d41c4a39-0c86-4628-b0ee-72fdbc640a87] Completed 200 OK in 1ms (Views: 0.1ms | Allocations: 110)
cosmos-openc3-traefik-1                   | 172.21.0.1 - - [09/Nov/2023:15:59:28 +0000] "GET / HTTP/1.1" 403 331 "-" "-" 15 "web-router@file" "http://openc3-minio:9000" 0ms
cosmos-openc3-traefik-1                   | 172.21.0.1 - - [09/Nov/2023:15:59:28 +0000] "GET /favicon.ico HTTP/1.1" 403 333 "-" "-" 16 "base-router@file" "http://openc3-minio:9000" 0ms
cosmos-openc3-cosmos-cmd-tlm-api-1        | I, [2023-11-09T15:59:33.003012 #19]  INFO -- : [9f737285-dad5-4bf4-9732-d36b1cef58ff] Started GET "/openc3-api/traefik" for 172.21.0.8 at 2023-11-09 15:59:33 +0000
cosmos-openc3-cosmos-cmd-tlm-api-1        | I, [2023-11-09T15:59:33.003682 #19]  INFO -- : [9f737285-dad5-4bf4-9732-d36b1cef58ff] Processing by MicroservicesController#traefik as HTML
cosmos-openc3-cosmos-cmd-tlm-api-1        | I, [2023-11-09T15:59:33.004263 #19]  INFO -- : [9f737285-dad5-4bf4-9732-d36b1cef58ff] Completed 200 OK in 0ms (Views: 0.1ms | Allocations: 110)
```

The issue may be with these lines in the logs pertaining to openc3-minio:9000 ? :

```bash
cosmos-openc3-traefik-1                   | 172.21.0.1 - - [09/Nov/2023:15:59:28 +0000] "GET / HTTP/1.1" 403 331 "-" "-" 15 "web-router@file" "http://openc3-minio:9000" 0ms
cosmos-openc3-traefik-1                   | 172.21.0.1 - - [09/Nov/2023:15:59:28 +0000] "GET /favicon.ico HTTP/1.1" 403 333 "-" "-" 16 "base-router@file" "http://openc3-minio:9000" 0ms
```

Is there an issue with this https://github.com/OpenC3/cosmos/blob/main/compose.yaml#L38 ?

haisamido, 2023-11-09:
I don't think it's related to the ssh tunnel, because if I do `curl localhost:2900` on the remote EC2 RHEL8 instance I also get:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>AccessDenied</Code><Message>Access Denied.</Message><Key>base/index.html</Key><BucketName>tools</BucketName><Resource>/tools/base/index.html</Resource><RequestId>179604C8C9354EF9</RequestId><HostId>dd9025bab4ad464b049177c95eb6ebf374d3b3fd1af9251148b658df7ac2e3e8</HostId></Error>
```

while on my laptop when I do `curl localhost:2900` to connect to my local running containers I get this

```html
<!doctype html><html lang="en"><head><meta charset="UTF-8"><meta name="viewport" content="width=device-width,initial-scale=1"><meta http-equiv="X-UA-Compatible" content="ie=edge"><title>OpenC3</title><link rel="stylesheet" href="/css/roboto.css"/><link rel="stylesheet" href="/css/materialdesignicons-7.3.67.min.css"/><link rel="stylesheet" href="/css/vuetify-2.7.1.min.css"/><link rel="icon" type="image/png" sizes="32x32" href="/favicon-32x32.png"><link rel="icon" type="image/png" sizes="16x16" href="/favicon-16x16.png"><meta name="theme-color" content="#ffffff"><script src="/js/regenerator-runtime-0.14.0.min.js"></script><meta http-equiv="Content-Security-Policy" content="default-src 'self' blob: data: https: http://localhost:* http://host.docker.internal:* http://*.local:*; script-src 'unsafe-inline' 'unsafe-eval' https: blob: http://localhost:* http://host.docker.internal:* http://*.local:*; connect-src blob: https: wss: http://localhost:* http://host.docker.internal:* http://*.local:* ws://localhost:* ws://host.docker.internal:* ws://*.local:*; style-src 'unsafe-inline' https: http://localhost:* http://host.docker.internal:* http://*.local:*; object-src 'none';"><meta name="importmap-type" content="systemjs-importmap"/><script type="systemjs-importmap">{
      "imports": {
        "single-spa": "/js/single-spa-5.9.5.min.js",
        "vue": "/js/vue-2.7.14.min.js",
        "vuetify": "/js/vuetify-2.7.1.min.js",
        "vue-router": "/js/vue-router-3.5.4.min.js",
        "vuex": "/js/vuex-3.6.2.min.js",
        "openc3-tool-base": "/openc3-tool-base.js"
      }
    }</script><link rel="preload" href="/js/single-spa-5.9.5.min.js" as="script"/><link rel="preload" href="/js/vue-2.7.14.min.js" as="script"/><link rel="preload" href="/js/vuetify-2.7.1.min.js" as="script"/><link rel="preload" href="/js/vue-router-3.5.4.min.js" as="script"/><link rel="preload" href="/js/vuex-3.6.2.min.js" as="script"/><script type="systemjs-importmap" src="/openc3-api/map.json"></script><script src="/js/auth.js"></script><script src="/js/import-map-overrides-3.1.0.min.js"></script><script src="/js/systemjs-6.14.2.min.js"></script><script src="/js/amd.min.js"></script><script src="/js/named-register.min.js"></script></head><body><div id="openc3-main"></div><import-map-overrides-full show-when-local-storage="devtools"></import-map-overrides-full><script>(function() {
    Promise.all([
      System.import('single-spa'),
      System.import('vue'),
      System.import('vuetify'),
      System.import('vue-router'),
      System.import('vuex'),
    ]).then(function (modules) {
      System.import('openc3-tool-base')
    })
  })()</script></body></html>
```

ryanmelt, 2023-11-09:
something is probably configured wrong on the RHEL install.   Seeing the docker logs on the init container would be helpful

haisamido, 2023-11-13:
> something is probably configured wrong on the RHEL install. Seeing the docker logs on the init container would be helpful

Interestingly the logs reveal `exec /openc3/init.sh failed: Permission denied` which is unusual! I've had this kind of issue of when running docker on linux hosts. 

```logs
cosmos-openc3-redis-ephemeral-1           | 1:M 13 Nov 2023 16:02:32.842 * monotonic clock: POSIX clock_gettime
cosmos-openc3-redis-ephemeral-1           | 1:M 13 Nov 2023 16:02:32.842 * Running mode=standalone, port=6380.
cosmos-openc3-redis-ephemeral-1           | 1:M 13 Nov 2023 16:02:32.842 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
cosmos-openc3-redis-ephemeral-1           | 1:M 13 Nov 2023 16:02:32.843 * Server initialized
cosmos-openc3-redis-ephemeral-1           | 1:M 13 Nov 2023 16:02:32.843 * Ready to accept connections tcp
cosmos-openc3-operator-1                  | {"time":1699891355548329991,"@timestamp":"2023-11-13T16:02:35.548+00:00","severity":"INFO","microservice_name":"MicroserviceOperator","container_name":"b7a68512f177","log":"OpenC3::MicroserviceOperator Monitoring processes every 5.0 sec...","type":"log"}
cosmos-openc3-cosmos-init-1               | [FATAL tini (6)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (7)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (6)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (7)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (7)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (7)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (6)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (6)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (6)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-init-1               | [FATAL tini (7)] exec /openc3/init.sh failed: Permission denied
cosmos-openc3-cosmos-script-runner-api-1  | => Booting Puma
cosmos-openc3-cosmos-script-runner-api-1  | => Rails 7.1.2 application starting in production 
cosmos-openc3-cosmos-script-runner-api-1  | => Run `bin/rails server --help` for more startup options
cosmos-openc3-cosmos-script-runner-api-1  | [6] Puma starting in cluster mode...
```

ryanmelt, 2023-11-13:
I made some changes in our 5.12.0 release to better handle podman, and permissions on various linux systems with restrictive umasks.   Give that a shot.

haisamido, 2023-11-13:
> I made some changes in our 5.12.0 release to better handle podman, and permissions on various linux systems with restrictive umasks. Give that a shot.

BTW, I'm using docker-ce and not podman.

ryanmelt, 2023-11-13:
The permissions stuff may help though.

closed 2023-12-07 by jmthomas