OpenC3/cosmos #935: operator and cmd-tlm-api container keeps restarting. Errno::ETIMEDOUT

avan989, 2023-11-15:
I am having a problem with the cosmos where the operator and cmd-tlm-API container keep restarting. When I look through the log, I see a Timeout on a socket is causing this. Do you have an idea of what is causing this?  

```
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:178:in `log_message'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:106:in `info'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:131:in `info'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/operators/operator.rb:358:in `run'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/operators/operator.rb:385:in `run'
	from microservice_operator.rb:195:in `<main>'
/usr/lib/ruby/3.2.0/socket.rb:65:in `connect_internal': Operation timed out - user specified timeout (Errno::ETIMEDOUT)
	from /usr/lib/ruby/3.2.0/socket.rb:141:in `connect'
	from /usr/lib/ruby/3.2.0/socket.rb:645:in `block in tcp'
	from /usr/lib/ruby/3.2.0/socket.rb:231:in `each'
	from /usr/lib/ruby/3.2.0/socket.rb:231:in `foreach'
	from /usr/lib/ruby/3.2.0/socket.rb:635:in `tcp'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client/ruby_connection.rb:117:in `connect'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client/ruby_connection.rb:49:in `initialize'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client.rb:688:in `new'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client.rb:688:in `block in connect'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client/middlewares.rb:12:in `connect'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client.rb:687:in `connect'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client.rb:674:in `raw_connection'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client.rb:641:in `ensure_connected'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-client-0.17.0/lib/redis_client.rb:237:in `call_v'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-5.0.7/lib/redis/client.rb:90:in `call_v'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-5.0.7/lib/redis.rb:167:in `block in send_command'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-5.0.7/lib/redis.rb:166:in `synchronize'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-5.0.7/lib/redis.rb:166:in `send_command'
	from /usr/lib/ruby/gems/3.2.0/gems/redis-5.0.7/lib/redis/commands/streams.rb:59:in `xadd'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/store_autoload.rb:186:in `block in write_topic'
	from /usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:110:in `block (2 levels) in with'
	from /usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `handle_interrupt'
	from /usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `block in with'
	from /usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `handle_interrupt'
	from /usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `with'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/store_autoload.rb:185:in `write_topic'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/store_autoload.rb:58:in `public_send'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/store_autoload.rb:58:in `method_missing'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/topics/topic.rb:29:in `public_send'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/topics/topic.rb:29:in `method_missing'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:210:in `block in log_message'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:178:in `synchronize'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:178:in `log_message'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:106:in `info'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/utilities/logger.rb:131:in `info'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/operators/operator.rb:358:in `run'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.11.0/lib/openc3/operators/operator.rb:385:in `run'
```

ryanmelt, 2023-11-15:
It can't connect to Redis.  Is your Redis container healthy?
```docker ps```

avan989, 2023-11-15:
The Redis container looks fine. I did notice that if I disabled the firewall, things seemed to work. I don't know if this is a network issue with docker.

```
9b2ab58d1648   openc3inc/openc3-cosmos-init:5.11.0                "/sbin/tini -- /open…"   About a minute ago   Up About a minute                                                                               cosmos-project-openc3-cosmos-init-1
91ce9fafb3c3   openc3inc/openc3-cosmos-script-runner-api:5.11.0   "/sbin/tini -- rails…"   About a minute ago   Up About a minute               2902/tcp                                                        cosmos-project-openc3-cosmos-script-runner-api-1
c293a9daf49d   openc3inc/openc3-operator:5.11.0                   "/sbin/tini -- ruby …"   About a minute ago   Up 6 seconds                    0.0.0.0:1235-1237->1235-1237/udp, :::1235-1237->1235-1237/udp   cosmos-project-openc3-operator-1
88fd8b093e56   openc3inc/openc3-cosmos-cmd-tlm-api:5.11.0         "/sbin/tini -- rails…"   About a minute ago   Restarting (1) 17 seconds ago                                                                   cosmos-project-openc3-cosmos-cmd-tlm-api-1
baba340746c3   openc3inc/openc3-traefik:5.11.0                    "/entrypoint.sh trae…"   About a minute ago   Up About a minute               127.0.0.1:2900->80/tcp, 127.0.0.1:2943->443/tcp                 cosmos-project-openc3-traefik-1
ef90e2401b7a   openc3inc/openc3-minio:5.11.0                      "/usr/bin/docker-ent…"   About a minute ago   Up About a minute               9000/tcp                                                        cosmos-project-openc3-minio-1
e95d26eb2dbd   openc3inc/openc3-redis:5.11.0                      "docker-entrypoint.s…"   About a minute ago   Up About a minute               3680/tcp, 6379/tcp                                              cosmos-project-openc3-redis-ephemeral-1
824b316fed7e   openc3inc/openc3-redis:5.11.0                      "docker-entrypoint.s…"   About a minute ago   Up About a minute               3680/tcp, 6379/tcp                                              cosmos-project-openc3-redis-1

```

ryanmelt, 2023-11-15:
Never seen that before.  So the redis container logs look fine?
```docker logs cosmos-project-openc3-redis-ephemeral-1```

avan989, 2023-11-16:
the container looks fine: 

```
1:C 15 Nov 2023 22:27:01.180 # WARNING Memory overcommit must be enabled! Without it, a background save or replication may fail under low memory condition. Being disabled, it can also cause failures without low memory condition, see https://github.com/jemalloc/jemalloc/issues/1328. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
1:C 15 Nov 2023 22:27:01.181 * oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
1:C 15 Nov 2023 22:27:01.181 * Redis version=7.2.1, bits=64, commit=00000000, modified=0, pid=1, just started
1:C 15 Nov 2023 22:27:01.181 * Configuration loaded
1:M 15 Nov 2023 22:27:01.183 * Increased maximum number of open files to 10032 (it was originally set to 1024).
1:M 15 Nov 2023 22:27:01.183 * monotonic clock: POSIX clock_gettime
1:M 15 Nov 2023 22:27:01.185 * Running mode=standalone, port=6380.
1:M 15 Nov 2023 22:27:01.189 * Server initialized
1:M 15 Nov 2023 22:27:01.189 * Ready to accept connections tcp
```

ryanmelt, 2023-11-16:
Do you have a support contract?  If so, we can do a call and debug.

avan989, 2023-11-16:
No unfortunately not at the moment. Maybe in the future.

closed 2023-12-08 by jmthomas