OpenC3/cosmos #937: Accessing logged packets from a script

Jesse-Millwood, 2023-11-16:
I would like to access logged telemetry from a script for a couple of reasons: 
- I would like to run a script in the script runner, issue some commands, monitor some telemetry, and archive the logged telemetry.
- I would like to process some logged telemetry to look for certain things like dropped packets and the like, which might be too intensive at runtime since we have a large amount of packets and checking the sequence number for all of them might get bogged down. 

Currently, I am printing out the start and end time of my script and using that in the Data Extractor so that I can archive the chunk of packets for a time frame. 

I tried digging into how the Data Extractor does this and it seems like it uses ActionCable to get a subscription? I'm fairly unfamiliar with web programming and this type of architecture so I'm having some trouble connecting the dots here. I do see the streaming api, is that available to scripts in the script runner? It seems like with the container architecture, the data I want would be available, I'm just having trouble getting to it.

It seems like ActionCable is bundled with rails? If that is true, is it already available to scripts running from the script runner? If it is, how would a ruby script use the ActionCable library? I tried looking for actioncable in ruby files in openc3 and tried copying some of the lines into a script to see if I could just access ActionCable and it seems

ryanmelt, 2023-11-16:
Not well documented yet, but you our looking to use the Stream Api through our script/websocket_api_code code:
https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/script/web_socket_api.rb

Something like this commented out example code from the bottom of the file:

```
OpenC3::StreamingWebSocketApi.new do |api|
  api.add(items: ['DECOM__TLM__INST__HEALTH_STATUS__TEMP1__CONVERTED', 'DECOM__TLM__INST__HEALTH_STATUS__TEMP2__CONVERTED'])
  5.times do
    puts api.read
  end
  api.remove(items: ['DECOM__TLM__INST__HEALTH_STATUS__TEMP1__CONVERTED'])
  5.times do
    puts api.read
  end
end

# Warning this saves all data to RAM. Do not use for large queries
data = OpenC3::StreamingWebSocketApi.read_all(items: ['DECOM__TLM__INST__HEALTH_STATUS__TEMP1__CONVERTED', 'DECOM__TLM__INST__HEALTH_STATUS__TEMP2__CONVERTED'], start_time: Time.now - 30, end_time: Time.now + 30)
```

closed 2024-10-12 by jmthomas

hmaclachlan, 2024-10-17:
@ryanmelt How would I configure this so that the packet would automatically download similar to how the packet downloads when you press "process" in Data Extractor?

hmaclachlan, 2024-11-07:
@ryanmelt 

Hello

I am looking to automate the process of processing packets from the Data Extractor so it can be performed within scripts that are being ran, based off of variables defined within that script (ie. start and end time). I am using the method described above, but it takes a while to add all rows of telemetry. Is there a better way to do this?

```
def download_tlm(tlm_items, test_start, test_end, headers, filepath)
  OpenC3::StreamingWebSocketApi.new do |api|
    api.add(items: `tlm_items)`
  end
  
  data = OpenC3::StreamingWebSocketApi.read_all(items: tlm_items, start_time: test_start, end_time: test_end)
  
  csv_content = CSV.generate do |csv|
    csv << headers
    data.each do |entry|
      row = [entry[tlm_items[0]]] + tlm_items[1..-1].map { |item| entry[item] }
      csv << row
    end
  end
  
  put_target_file(filepath, csv_content)
  download_file(filepath)
end
```

hmaclachlan, 2024-11-12:
@ryanmelt 
Hello
I am looking to automate processing packets from the Data Extractor so it can be performed within scripts that are being ran, based off of variables defined within that script (ie. start and end time). I am using the method described above and adding the items to a .csv, but it takes a while to add all rows of telemetry. Is there a better way to do this? I am unable to locate any code that details how the Data Extractor performs on the backend. I tried using batch/parallel processing methods to speed up this process, but COSMOS does not seem to recognize the 'concurrent' library.

```
def download_tlm(tlm_items, test_start, test_end, headers, filepath)
  OpenC3::StreamingWebSocketApi.new do |api|
    api.add(items: tlm_items)
  end
  
  data = OpenC3::StreamingWebSocketApi.read_all(items: tlm_items, start_time: test_start, end_time: test_end) # Warning this saves all data to RAM. Do not use for large queries
  
  csv_content = CSV.generate do |csv|
    csv << headers
    data.each do |entry|
      row = [entry[tlm_items[0]]] + tlm_items[1..-1].map { |item| entry[item] }
      csv << row
    end
  end
  
  put_target_file(filepath, csv_content)
  download_file(filepath)
end
```

```
def download_tlm(tlm_items, test_start, test_end, headers, filepath)
  OpenC3::StreamingWebSocketApi.new do |api|
    api.add(items: tlm_items)
  end

  data = OpenC3::StreamingWebSocketApi.read_all(items: tlm_items, start_time: test_start, end_time: test_end)

  # Use a thread pool for parallel processing
  pool = Concurrent::FixedThreadPool.new(4)
  csv_content = CSV.generate do |csv|
    csv << headers
    data.each_slice(1000) do |batch|
      futures = batch.map do |entry|
        Concurrent::Future.execute(executor: pool) do
          [entry[tlm_items[0]]] + tlm_items[1..-1].map { |item| entry[item] }
        end
      end
      futures.each { |future| csv << future.value }
    end
  end

  pool.shutdown
  pool.wait_for_termination

  put_target_file(filepath, csv_content)
  download_file(filepath)
end
```