OpenC3/cosmos #941: Python standalone script

omermesser, 2023-11-19:
I have openc3 set up and running on 172.0.0.1:2900 and I can successfully send commands and view packets manually
I have read the scripting documentations.
I do not understand how I can send commands the receive them using the openc3 package in python.
I want to be able to import cmd and tlm and use it in a standalone python script.
this is my sending cmd code:
```
import os
import sys
from openc3.script import *

# See openc3/docs/environment.md for environment documentation

os.environ["OPENC3_API_PASSWORD"] = "admin"
os.environ["OPENC3_LOG_LEVEL"] = "DEBUG"
os.environ["OPENC3_API_SCHEMA"] = "http"
os.environ["OPENC3_API_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_API_PORT"] = "2900"

res = cmd('BLOCKONE', 'MIC_LSL-MIC_CSP_PING_RES', {'MICRON_ID': 40})
print(res)
```
I get this error:
```
Exception has occurred: JsonDRbError
No response from server: Request: {'url': 'http://openc3-cosmos-cmd-tlm-api:2901/openc3-api/api', 'data': '{"jsonrpc": "2.0", "method": "cmd", "params": ["BLOCKONE", "MIC_LSL-MIC_CSP_PING_RES", {"MICRON_ID": 40}], "keyword_params": {"timeout": null, "scope": "DEFAULT"}, "id": 0}', 'headers': {'User-Agent': 'OpenC3 / v5 (ruby/openc3/lib/io/json_drb_object)', 'Content-Type': 'application/json-rpc'}} ::: None ::: Exception: ConnectionError(MaxRetryError("HTTPConnectionPool(host='openc3-cosmos-cmd-tlm-api', port=2901): Max retries exceeded with url: /openc3-api/api (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000027AC18B4950>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))"))
  File "C:\COSMOS_V5\pes-development\BLOCKONE\procedures\MIC_PING\TEST.py", line 13, in <module>
    res = cmd('BLOCKONE', 'MIC_LSL-MIC_CSP_PING_RES', {'MICRON_ID': 40})
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openc3.io.json_drb_object.JsonDRbError: No response from server: Request: {'url': 'http://openc3-cosmos-cmd-tlm-api:2901/openc3-api/api', 'data': '{"jsonrpc": "2.0", "method": "cmd", "params": ["BLOCKONE", "MIC_LSL-MIC_CSP_PING_RES", {"MICRON_ID": 40}], "keyword_params": {"timeout": null, "scope": "DEFAULT"}, "id": 0}', 'headers': {'User-Agent': 'OpenC3 / v5 (ruby/openc3/lib/io/json_drb_object)', 'Content-Type': 'application/json-rpc'}} ::: None ::: Exception: ConnectionError(MaxRetryError("HTTPConnectionPool(host='openc3-cosmos-cmd-tlm-api', port=2901): Max retries exceeded with url: /openc3-api/api (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000027AC18B4950>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))"))
```

And I cannot use tlm functions.

ryanmelt, 2023-11-19:
Remove all the os.environ calls.  Those are set for you if running in scriptrunner.

ryanmelt, 2023-11-19:
If you are outside of Scriptrunner then the os.environ lines need to be before the imports

omermesser, 2023-11-19:
> If you are outside of Scriptrunner then the os.environ lines need to be before the imports

I want to execute this outside of scriptrunner.
I have updated to this:
```
import os
import sys
os.environ["OPENC3_API_PASSWORD"] = "admin"
os.environ["OPENC3_LOG_LEVEL"] = "DEBUG"
os.environ["OPENC3_API_SCHEMA"] = "http"
os.environ["OPENC3_API_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_API_PORT"] = "2900"

from openc3.script import *

res = cmd('BLOCKONE', 'MIC_LSL-MIC_CSP_PING', {'MICRON_ID': 40})
print(res)
```

and receive another error:
```
Exception has occurred: ConnectionError
Error 11001 connecting to openc3-redis-ephemeral:6380. getaddrinfo failed.
socket.gaierror: [Errno 11001] getaddrinfo failed

During handling of the above exception, another exception occurred:

  File "C:\COSMOS_V5\pes-development\BLOCKONE\procedures\MIC_PING\TEST.py", line 11, in <module>
    res = cmd('BLOCKONE', 'MIC_LSL-MIC_CSP_PING', {'MICRON_ID': 40})
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
redis.exceptions.ConnectionError: Error 11001 connecting to openc3-redis-ephemeral:6380. getaddrinfo failed.
```

omermesser, 2023-11-19:
I have managed to make it work by adding these:
```
os.environ["OPENC3_REDIS_EPHEMERAL_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_REDIS_EPHEMERAL_PORT"] = "6380"
os.environ["OPENC3_REDIS_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_REDIS_PORT"] = "6379"
os.environ["OPENC3_REDIS_USERNAME"] = "openc3"
os.environ["OPENC3_REDIS_PASSWORD"] = "openc3password"
```

But I have another issue. I dont have `tlm()` function in my imports

omermesser, 2023-11-19:
I have another issue. how can I use `tlm()` functions? I dont have them in the import

omermesser, 2023-11-19:
I have managed to solve it by adding these to the environ
```
os.environ["OPENC3_REDIS_EPHEMERAL_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_REDIS_EPHEMERAL_PORT"] = "****"
os.environ["OPENC3_REDIS_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_REDIS_PORT"] = "****"
os.environ["OPENC3_REDIS_USERNAME"] = "*****"
os.environ["OPENC3_REDIS_PASSWORD"] = "*****"
```
But now I dont have the `tlm()` function. how can I read the packets?

jmthomas, 2023-11-19:
This worked for me:
```
import os

# See openc3/docs/environment.md for environment documentation
os.environ["OPENC3_API_PASSWORD"] = "password"
os.environ["OPENC3_LOG_LEVEL"] = "DEBUG"
os.environ["OPENC3_API_SCHEMA"] = "http"
os.environ["OPENC3_API_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_API_PORT"] = "2900"
# This import must be after the os.environ settings
from openc3.script import *

print(tlm("INST HEALTH_STATUS TEMP1"))
```

omermesser, 2023-11-20:
> This worked for me:
>
> ```
> import os
>
> # See openc3/docs/environment.md for environment documentation
> os.environ["OPENC3_API_PASSWORD"] = "password"
> os.environ["OPENC3_LOG_LEVEL"] = "DEBUG" [...]

I have used
`from openc3.api.tlm_api import *`
and it solved it. is it good practice? otherwise I don't have access to the tlm functions

jmthomas, 2023-11-20:
For external scripts you should be using the `openc3.script` library, not the `openc3.api` library. This is all very new python support. What version of COSMOS are you running?

omermesser, 2023-11-21:
I'm using the latest in version in PyPI:
openc3                    5.12.0

omermesser, 2023-11-22:
This is showing what I'm trying to do
[image]

jmthomas, 2023-11-24:
You don't have the password, hostname or port set:
```
import os

# See openc3/docs/environment.md for environment documentation
os.environ["OPENC3_API_PASSWORD"] = "password"
os.environ["OPENC3_API_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_API_PORT"] = "2900"
# This import must be after the os.environ settings

from openc3.script import *
cmd("INST ABORT")
print(tlm("INST HEALTH_STATUS TEMP1"))
```

omermesser, 2023-11-26:
Hey thanks for replying but actually I do have them.
This is what im setting up before importing:
```
os.environ["OPENC3_API_PASSWORD"] = "admin"
os.environ["OPENC3_LOG_LEVEL"] = "INFO"
os.environ["OPENC3_API_SCHEMA"] = "http"
os.environ["OPENC3_API_HOSTNAME"] = "127.0.0.1"
os.environ["OPENC3_API_PORT"] = "2900"
os.environ["OPENC3_REDIS_EPHEMERAL_HOSTNAME"] = "*****"
os.environ["OPENC3_REDIS_EPHEMERAL_PORT"] = "******"
os.environ["OPENC3_REDIS_HOSTNAME"] = "*****"
os.environ["OPENC3_REDIS_PORT"] = "*****"
os.environ["OPENC3_REDIS_USERNAME"] = "*****"
os.environ["OPENC3_REDIS_PASSWORD"] = "*****"
```

omermesser, 2023-11-29:
alright, thanks for trying.
I still can't manage to get access to the `tlm` functions from `openc3.script`
is there any more information I can provide that will help you?

jmthomas, 2023-11-29:
If you have an Enterprise License or support contract you can setup a call here: https://www.openc3.com/demo