OpenC3/cosmos #942: I'm trying to upload a file using a command that sends chunks of binary file

omermesser, 2023-11-19:
One of the parameters of the cmd is the binary data itself.
Another parameter is the data length (in bytes).
I'm reading the file and sending the cmd with chunks of bytes multiple times (the chunk is 1754 bytes).

the data of a chunk for example is:
```
b '\xdd\x00\x00\x00D3"\x11\x88wfU\xcc\xbb\xaa\x99fU\x99\xaaXNLX\x00\x00\x00\x00\x80\x0f...'
```

(the 3 dots in the example above is my change so that the issue will not be long)

now I'm getting this and I have error saying the data sent is longer than specified:
```
Exception has occurred: RuntimeError
value of 2496 bytes does not fit into 1754 bytes for data_type BLOCK
.
. (call stack)
.
RuntimeError: value of 2496 bytes does not fit into 1754 bytes for data_type BLOCK
```

I have opened an issue in COSMOS4 and was told to move to COSMOS 5 so I did.
[https://github.com/BallAerospace/python-ballcosmos/issues/30](url)

**How should I handle the bytes? it seems openc3 python API stringifys them**

jmthomas, 2023-11-19:
How are you sending the command? This works for me in the demo:
```
data = b"\x00\x01\x02\x03"
cmd("INST", "MEMLOAD", {"DATA": data})
```

omermesser, 2023-11-20:
> How are you sending the command? This works for me in the demo:
> 
> ```
> data = b"\x00\x01\x02\x03"
> cmd("INST", "MEMLOAD", {"DATA": data})
> ```

I'm sending the command exactly like the example you gave:
```
cmd_name = "MY_FILE_UPLOAD"
cmd_params = {
    "MY_ID": my_id,
    "DATA_FILE_ID": data_file_id,
    "DATA_ENTRY_ID": data_entry_id,
    "DATA_OFFSET": 0, # fixed
    "DATA_LEN": data_len,
    "DATA_BLOCK": data_block
}
cmd('MyTarget', cmd_name, cmd_params)
```

I have bytes that cannot be represented by UTF-8 or ASCII.

So after sending this command im receiving the error.

omermesser, 2023-11-20:
I think is it because of the function that converts bytes to string,
in the file Python\Lib\site-packages\openc3\io\json_rpc.py in line 325:
```
def _convert_bytearray_to_string_raw(object_):
    if isinstance(object_, (bytes, bytearray)):
        return object_.decode("latin-1")
    if isinstance(object_, dict):
        for key, value in object_.items():
            object_[key] = _convert_bytearray_to_string_raw(value)
        return object_
    if isinstance(object_, (tuple, list)):
        object_ = list(object_)
        index = 0
        for value in object_:
            object_[index] = _convert_bytearray_to_string_raw(value)
            index += 1
        return object_
    return object_
```
it decodes the bytes to latin-1 so maybe its there?

jmthomas, 2023-11-20:
@omermesser  thank you for this bug report! I think I have it fixed in the linked PR.

omermesser, 2023-11-21:
That's great!
I would like to check it but I'm a bit new to git.

I'm currently getting access to the python code via PyPI by running
`pip install openc3`

what is the correct practice to get access the the branch you created which will be similar to having a package installed?

omermesser, 2023-11-22:
It is working flawlessly! Thank you for supporting!
You may merge it to main so we can use `pip install openc3` in the future

PR #946 (2023-11-22): Fix python encoding of binary
closes #942
closed 2023-11-22 by omermesser