OpenC3/cosmos #948: Default Seccomp Profile

CalebRoden, 2023-11-22:
Im having trouble running the initial install of the tutorial repository

I have updated docker, made sure compose is running correctly, however I am getting the error 

"WARNING: Daemon is not using the default seccomp profile"

So I went to the docker website and downloaded the default.json file, however Im unsure of what to do next.
Let me know if anyone has found a workaround or a setting I need to change

Im using a Macbook Pro with the intel chip on Sonoma 14.1.1

Thanks much

jmthomas, 2023-11-22:
I've never seen that before but perhaps start your investigation here: https://docs.docker.com/engine/security/seccomp/

ryanmelt, 2023-11-23:
I always get that same message on my mac and it doesn't hurt anything.
The real error is probably something different.  Can you copy the full shell output?
Or run "docker ps" and make sure all the container are running successfully

closed 2024-01-12 by jmthomas