OpenC3/cosmos #952: Connecting to and getting data from an existing API

MahimaSeth, 2023-11-27:
Hi all,

I'm a pretty junior developer tasked with getting stuff working in OpenC3!
Sorry if my questions are answered already/in the guide, I need a little more direction and details on some "basics" as it looks like the guides are written for a more experienced user.

Here are my questions,  I would be grateful for some help :) Please let me know if I can clarify something I've said
1. I want openC3 to send a simple get request to an API that can be accessed with a link that you can copy and paste into a browser.  To begin with, I don't care about accessing all the different elements in the json file that the API will send back. I just want to get the two talking and for openC3 to simply show me the json file that is received.  

I have already looked here: https://docs.openc3.com/docs/development/json-api  
...and that isn't helping me. I've also tried to follow the debugging section. 

2. How can start dealing with all the info in the json file? I mistakenly? tried to handle the json file by treating it no different to telemetry being sent back (in the tlm.txt file, with $key notation)...failed miserably because I tried some weird way of putting the link into the plugin.txt file because I couldn't get OpenC3 talking to the API in the first place.........

Any help on this would be much appreciated. I'm use to seeing fairly detailed examples (having written in python mostly!), but there isn't much of that available to look at, or if there is, please point me to it!

Thanks!

ryanmelt, 2023-11-27:
You're going to want to use the new HttpClientInterface.  The documentation for that isn't done yet, but there is a good example here:
https://github.com/OpenC3/cosmos/tree/main/openc3-cosmos-init/plugins/packages/openc3-cosmos-http-example

MahimaSeth, 2023-12-06:
Hey @ryanmelt thanks for pointing me to that. It was very helpful. 
I've managed to connect to the REST endpoint (CmdTlmServer is showing HTTP_CLIENT_INT and HTTP_SERVER_INT) are both "green" and connected).
However, I'm unsure as to why I am not seeing any telemetry come through. I've tried sending a command through command sender: gives a "Commands cannot be sent to HttpServerInterface" (screenshot below: Picture #1)..unsure why? I thought that you have to SEND a GET request (command) to get the telemetry back??

>Additionally, I have attached picture#2 , which is a screenshot of the cmd_tlm.txt for the API output (server)
>picture #3, which is a screenshot of cmd_tlm.txt for the client (which I guess is OpenC3?...but I have called it "xxx Ground station)

Since there is no documentation for this yet, I would greatly appreciate some direction on why there might be no telemetry coming through!

Thanks!
[image]
[image]
[image]

ryanmelt, 2023-12-07:
An HttpServerInterface can only receive requests (Its like a web server).   That's why you can't send commands.
For sending commands you will need to use the HttpClient.

jmthomas, 2025-12-31:
This is now documented: https://docs.openc3.com/docs/configuration/interfaces#http-client-interface

closed 2025-12-31 by jmthomas