OpenC3/cosmos #961: Troubleshooting COSMOS and cFS for Ubuntu 22.04

schuuust, 2023-11-29:
I  implemented cFS with COSMOS using the [official guide here](https://docs.openc3.com/docs/guides/cfs) and found some information that appears to out of date and required a bit of troubleshooting to resolve. These are my solutions to the issues I encountered and how I resolved them, perhaps it'll benefit someone else as well! 

I ran through these instructions with the following:

- Ubuntu 22.04
- Docker (24.0.5)
- OpenC3 COSMOS (5.12.0)
- NASA cFS (Aquila - 6.7.0)

**Docker Ubuntu Image**

(In the *Create Dockerfile in cFS dir* section)

First issue I encountered is in the Dockerfile for cFS. The suggested `ubuntu:22.10` docker image does not seem to exist any more.
I changed to use the `ubuntu:22.04` image by replacing both instances of `ubuntu:22.10` with `ubuntu:22.04` in the Dockerfile for cFS.

**Expose cFS Docker Port 1234 as UDP**

(In the *Build and Run cFS* section)

By default, my docker image for cFS exposed port 1234 as tcp/ip, which seemed to prevent communications with Cosmos as it was configured to look for cFS over a UDP port. My solution for this was adding a flag to expose the cFS docker container's port over udp by adding the `--expose 1234/udp` flag.

Full command I used to run cFS:

`docker run --rm --cap-add CAP_SYS_RESOURCE --net=openc3-cosmos-network --name cfs -p1234:1234 -p1235:1235 --expose 1234/udp cfs`

*Side note:* I also ran this with the `--rm` flag. This automatically removes the container once it is brought down. This makes troubleshooting or repeatedly bringing it up and down a little smoother since I don't have to manually remove it every time. This might just be a personal preference thing.

**Potential Typo in Plugin Build Instructions**

(In the *Creating a COSMOS plugin for TM/TC interface with cFS* section)

The following instructions ask the user in the comment to go to the location of the cfs dir. Based on context, I assume the user should actually be in the openc3 dir? 

```bash
# cd .. to the location of the cfs dir
$PATH_TO_OPENC3/openc3.sh cli generate plugin CFS
```

**In Conclusion**

The original docs are explicitly for a different OS and version of Cosmos, but should anyone else be trying to implement them on Ubuntu with newer versions, maybe this info'll help! I'm not too hot with networking stuff so it took me a while to figure out haha

schuuust, 2023-11-30:
Thanks for the response!

Looks like the errors I'm having with pulling 22.10 are just local issues, I'll keep working on that.

Can confirm that `-p1234:1234/udp` works on my system!

Ah I see what the path comment meant now, I was misinterpreting the instructions.

jmthomas, 2023-11-30:
@schuuust did you need to add `/udp` to the 1235 port as well? e.g. `-p1235:1235/udp`

schuuust, 2023-12-01:
No, I did not. In fact, port 1235 can't be exposed as udp on both Cosmos and cFS, it causes a "port already in use" error in whichever one is launched second

closed 2023-12-12 by jmthomas