OpenC3/cosmos #968: Telemetry JSON API

rmslick, 2023-12-11:
I am trying to issue a request to get a full telem packet for a telem type on a target using the JSON API.

It is stated in the docs that "JSON API methods correspond to the COSMOS scripting API methods documented in the [Scripting Writing Guide](https://docs.openc3.com/docs/guides/script-writing)." but I am not having success accessing the get_tlm_packet method and it is not clear to me how to do so. 

Here is my attempted curl followed by the error msg I get (Can post more of the backtrace if needed):

curl -d '{"jsonrpc": "2.0", "method": "get_tlm_packet", "params": ["INST HEALTH_STATUS"], "id": 2, "keyword_params":{"scope":"DEFAULT"}}' http://localhost:2900/openc3-api/api  -H "Authorization: password"  

Error: arguments (given 1, expected 2)"

jmthomas, 2023-12-11:
You're really close but I think you're missing the scope keyword param. Please check out our documentation: https://docs.openc3.com/docs/development/curl

rmslick, 2023-12-11:
Okay that comma works! Thanks. But to me the convention isn't clear. I just tried to follow the convention of passing what seems like multiple (three) params in the case of the API call to get a specific data point. It seems like from that you could just remove the specific point within the packet to get the full packet.  To give a comma separated list for other api methods seems inconsistent or maybe I am just missing obvious documentation somewhere.

jmthomas, 2023-12-11:
You are correct. I'm currently working on a PR to make it take either two separate params or one space delimited param. Most of our APIs take both but there are few that currently only allow separate parameters.

closed 2023-12-11 by jmthomas