OpenC3/cosmos #975: Building widgets

rmslick, 2023-12-13:
I followed these steps to build a widget in a plugin directory with these steps outlined in this response: https://github.com/OpenC3/cosmos/issues/740#issuecomment-1629684699 which says to build by:
```
# Make sure openc3.sh is in your path
openc3.sh cli generate plugin testing
cd openc3-cosmos-testing
openc3.sh cli generate tool_vue mytool
yarn
rake build VERSION=1.0.0
```

But if I try to build the plugin with the openc3.sh script it fails with this trace. Can the widgets not be built using the openc3.sh script? Sorry if I am missing a guide in the docs somewhere explaining this. THanks!

./openc3.sh cli rake build VERSION=1.0.0 --trace
Error response from daemon: network with name openc3-cosmos-network already exists
** Invoke build (first_time)
** Invoke require_version (first_time)
** Execute require_version
** Execute build
yarn run build
rake aborted!
yarn run build failed
/openc3/local/rakefile:21:in `block (2 levels) in <top (required)>'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/file_utils.rb:57:in `sh'
/openc3/local/rakefile:20:in `block in <top (required)>'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:281:in `block in execute'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:281:in `each'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:281:in `execute'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:199:in `synchronize'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:199:in `invoke_with_call_chain'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/task.rb:188:in `invoke'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:160:in `invoke_task'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:116:in `block (2 levels) in top_level'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:116:in `each'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:116:in `block in top_level'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:125:in `run_with_threads'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:110:in `top_level'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:83:in `block in run'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:186:in `standard_exception_handling'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/lib/rake/application.rb:80:in `run'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.0.6/exe/rake:27:in `<top (required)>'
/usr/bin/rake:25:in `load'
/usr/bin/rake:25:in `<main>'
Tasks: TOP => build

ryanmelt, 2023-12-13:
Try running "yarn build" and see what errors that gets

rmslick, 2023-12-13:
yarn build appears to work fine too:

  File                                   Size                                                             Gzipped

  tools\log_messages_widget\js\app.js    427.72 KiB                                                       81.30 KiB
  tools\log_messages_widget\js\870.js    50.77 KiB                                                        17.51 KiB
  tools\log_messages_widget\js\662.js    14.87 KiB                                                        4.68 KiB
  tools\log_messages_widget\js\610.js    0.50 KiB                                                         0.34 KiB

  Images and other types of assets omitted.
  Build at: 2023-12-13T22:18:22.167Z - Hash: 543d55fcc1d2cb10 - Time: 15548ms

 DONE  Build complete. The tools\log_messages_widget directory is ready to be deployed.
 INFO  Check out deployment instructions at https://cli.vuejs.org/guide/deployment.html

ryanmelt, 2023-12-14:
Oh.  I missed the detail from your first post that you were trying to do this with openc3.sh cli rake.
Can't do that for node projects because the openc3-operator container does not container the node build environment.

Quick fix would be to edit your openc3.sh script and change the openc3-operator references to openc3-node

rmslick, 2023-12-14:
So making those changes to openc3.sh and re running the  build with clie, I get the following:

Digest: 
Status: Downloaded newer image for openc3inc/openc3-node:5.11.3
ruby: No such file or directory -- /openc3/bin/openc3cli (LoadError)

ryanmelt, 2023-12-18:
The node image gained ruby in 5.12.0.  So it should work on that version and going forward.

closed 2023-12-18 by ryanmelt