OpenC3/cosmos #990: Data of type "FLOAT" is auto-converted to "INT" if there are no non-zero decimals

GTI-Dakota, 2023-12-18:
**Describe the bug**
If a user has a command definition with a parameter of type "FLOAT", and does not include a non-zero decimal in that parameter when sending the command, COSMOS will turn the value into an int. This is problematic for a protocol that utilizes CBOR within it, because the encoded type will be set an "int", rather than the appropriate "float". As of currently, workaround code is required to fix this (See below). Not sure if this is fixed in later COSMOS versions.

**To Reproduce**
with parameter name "timeout", type FLOAT, value -1.01:
cmd("SYSTEM START_TEST with INTERFACE 'my_interface', TIMEOUT -1.01")

with parameter name "timeout", type FLOAT, value -1.0:
cmd("SYSTEM START_TEST with INTERFACE 'my_interface', TIMEOUT -1")

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
- OS: Linux Ubuntu
- OpenC3 COSMOS Version: 5.8.1
- Browser: N/A

jmthomas, 2023-12-19:
@GTI-Dakota would you have an opportunity to try this on the current version? If not it seems like you've identified a workaround in your protocol, correct?

GTI-Dakota, 2023-12-19:
The workaround is good to go for my protocol, but it's a bit of a kludge. In using this method, every packet being sent out must be checked via an API call (get_command) and run a conversion if any data types are specified as "FLOAT".

PR #1006 (2024-01-07): Make json, cbor, xml, html enforce data types
closes #990
closed 2024-01-07 by jmthomas