OpenC3/cosmos #994: ADD type to get_packets()

omermesser, 2023-12-20:
In Python I'm trying to use get_packets() and return the packets as CONVERTED.
the issue is that it always returns the packets as RAW.
```
id1 = subscribe_packets([['MyTarget','MyPacket']]) # subscribing
cmd('MyTarget','MyCommand', MyParams) # seding command
id, packets = get_packets(id1, block=2, block_delay=0.05) # retrieving packets
# packets are always RAW
```

I cannot use get_tlm_packet() because I send rapid commands and the last one will overwrite the true packet I wish to read.
If there is already a way of getting the packet as CONVERTED please let me know.

jmthomas, 2023-12-20:
Did you mean the opposite: All the packets are always CONVERTED? Because when I run this I only get converted packets which makes sense because in [tlm_api.py](https://github.com/OpenC3/cosmos/blob/54c57ce91757442c05167d54d760cd883193a80d/openc3/python/openc3/api/tlm_api.py#L408) we're always subscribed to the DECOM stream.

jmthomas, 2023-12-20:
Actually I think the confusion is that we're returning the fully decommutated packet which is all the types: RAW, CONVERTED, FORMATTED and WITH_UNITS. Here's an example script from the demo that shows what I mean:
```
id1 = subscribe_packets([['INST','HEALTH_STATUS']]) # subscribing
wait(2)
id, packets = get_packets(id1, block=2, block_delay=0.05) # retrieving packets
print(packets[0]['TEMP1'])
print(packets[0]['TEMP1__C'])
print(packets[0]['TEMP1__F'])
print(packets[0]['TEMP1__U'])
```

You need to explicitly add the type to the item where `__C` is CONVERTED. RAW is the default which is probably what you're seeing.

omermesser, 2023-12-21:
You are correct, I wasn't paying attention to the suffix of the keys. Thank you!

closed 2023-12-21 by omermesser