OpenC3/cosmos #998: inject_tlm throws "Auth error" inside processor, not in script

scandey, 2023-12-21:
**Describe the bug**
`inject_tlm` called from inside the `call()` method of a `Processor` class throws an error somehow related to authentication (openc3_token related maybe?) This exact code has worked in previous versions (currently have 5.9.1 running), though it also did not work in 5.12.0 tested earlier today. 

The same call from the scripting interface does not throw any errors. 

**To Reproduce**
Processor with `inject_tlm` throws error attached below. I'm working on a shorter reproducible case but it is proving to be more difficult than I anticipated. 

**Expected behavior**
Inject telemetry continues to work inside processors as in previous versions. 

**Environment (please complete the following information):**

- OS: MacOS 13.6.2 on Docker Desktop 4.25.2
- OpenC3 COSMOS Version 5.13.0
- Browser: Safari

```
MISSION_INT: AuthError : Token is required
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/utilities/authorization.rb:44:in `authorize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/api/tlm_api.rb:123:in `inject_tlm'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb.rb:268:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb.rb:268:in `process_request'
/src/app/controllers/api_controller.rb:88:in `handle_post'
/src/app/controllers/api_controller.rb:42:in `block in api'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/src/app/controllers/api_controller.rb:28:in `api'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/basic_implicit_render.rb:6:in `send_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/base.rb:224:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/rendering.rb:165:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/callbacks.rb:259:in `block in process_action'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/callbacks.rb:101:in `run_callbacks'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/callbacks.rb:258:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/rescue.rb:25:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/instrumentation.rb:74:in `block in process_action'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/notifications.rb:206:in `block in instrument'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/notifications/instrumenter.rb:58:in `instrument'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/notifications.rb:206:in `instrument'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/instrumentation.rb:73:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/params_wrapper.rb:261:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/base.rb:160:in `process'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal.rb:227:in `dispatch'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal.rb:309:in `dispatch'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/routing/route_set.rb:49:in `dispatch'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/routing/route_set.rb:32:in `serve'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:51:in `block in serve'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:131:in `block in find_routes'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:124:in `each'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:124:in `find_routes'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:32:in `serve'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/routing/route_set.rb:882:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/etag.rb:29:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/conditional_get.rb:43:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/head.rb:15:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/callbacks.rb:29:in `block in call'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/callbacks.rb:101:in `run_callbacks'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/callbacks.rb:28:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/debug_exceptions.rb:29:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/show_exceptions.rb:31:in `call'
/usr/lib/ruby/gems/3.2.0/gems/railties-7.1.2/lib/rails/rack/logger.rb:37:in `call_app'
/usr/lib/ruby/gems/3.2.0/gems/railties-7.1.2/lib/rails/rack/logger.rb:24:in `block in call'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:135:in `block in tagged'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:39:in `tagged'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:135:in `tagged'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/broadcast_logger.rb:240:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/railties-7.1.2/lib/rails/rack/logger.rb:24:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/remote_ip.rb:92:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/request_id.rb:28:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/runtime.rb:24:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/executor.rb:14:in `call'
[... 17 lines cut ...]
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb_object.rb:72:in `block in method_missing'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb_object.rb:61:in `synchronize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb_object.rb:61:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/script/script.rb:282:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/script/telemetry.rb:50:in `inject_tlm'
/tmp/d20231221-3657-i47tpa/targets/SPANI/lib/swem_event_message_processor.rb:96:in `call'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet.rb:1155:in `block in process'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet.rb:1154:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet.rb:1154:in `process'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet.rb:309:in `block in buffer='
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/structure.rb:524:in `block in synchronize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/structure.rb:524:in `synchronize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/structure.rb:524:in `synchronize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet.rb:302:in `buffer='
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/telemetry.rb:269:in `identify!'
/tmp/d20231221-3657-i47tpa/targets/ICEB/lib/interfaces/debundle_protocol.rb:63:in `read_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/interfaces/interface.rb:292:in `block (2 levels) in read'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/interfaces/interface.rb:291:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/interfaces/interface.rb:291:in `block in read'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/interfaces/interface.rb:251:in `loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/interfaces/interface.rb:251:in `read'
interface_microservice.rb:473:in `run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/microservices/microservice.rb:55:in `run'
interface_microservice.rb:691:in `<main>'
```

scandey, 2023-12-21:
Minimal reproducible case, including processor that injects telemetry. Install gem and run procedure 'inject.rb' to trigger the below error.
[openc3-cosmos-test 0_0_6.zip](https://github.com/OpenC3/cosmos/files/13736536/openc3-cosmos-test.0_0_6.zip)

```
Decom error: AuthError : Token is required
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/utilities/authorization.rb:44:in `authorize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/api/tlm_api.rb:123:in `inject_tlm'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb.rb:268:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/io/json_drb.rb:268:in `process_request'
/src/app/controllers/api_controller.rb:88:in `handle_post'
/src/app/controllers/api_controller.rb:42:in `block in api'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/src/app/controllers/api_controller.rb:28:in `api'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/basic_implicit_render.rb:6:in `send_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/base.rb:224:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/rendering.rb:165:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/callbacks.rb:259:in `block in process_action'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/callbacks.rb:101:in `run_callbacks'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/callbacks.rb:258:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/rescue.rb:25:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/instrumentation.rb:74:in `block in process_action'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/notifications.rb:206:in `block in instrument'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/notifications/instrumenter.rb:58:in `instrument'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/notifications.rb:206:in `instrument'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/instrumentation.rb:73:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal/params_wrapper.rb:261:in `process_action'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/abstract_controller/base.rb:160:in `process'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal.rb:227:in `dispatch'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_controller/metal.rb:309:in `dispatch'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/routing/route_set.rb:49:in `dispatch'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/routing/route_set.rb:32:in `serve'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:51:in `block in serve'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:131:in `block in find_routes'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:124:in `each'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:124:in `find_routes'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/journey/router.rb:32:in `serve'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/routing/route_set.rb:882:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/etag.rb:29:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/conditional_get.rb:43:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/head.rb:15:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/callbacks.rb:29:in `block in call'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/callbacks.rb:101:in `run_callbacks'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/callbacks.rb:28:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/debug_exceptions.rb:29:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/show_exceptions.rb:31:in `call'
/usr/lib/ruby/gems/3.2.0/gems/railties-7.1.2/lib/rails/rack/logger.rb:37:in `call_app'
/usr/lib/ruby/gems/3.2.0/gems/railties-7.1.2/lib/rails/rack/logger.rb:24:in `block in call'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:135:in `block in tagged'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:39:in `tagged'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:135:in `tagged'
/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.2/lib/active_support/broadcast_logger.rb:240:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/railties-7.1.2/lib/rails/rack/logger.rb:24:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/remote_ip.rb:92:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/request_id.rb:28:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/runtime.rb:24:in `call'
/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.2/lib/action_dispatch/middleware/executor.rb:14:in `call'
/usr/lib/ruby/gems/3.2.0/gems/rack-3.0.8/lib/rack/sendfile.rb:114:in `call'
[... cut ...]
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/topics/topic.rb:29:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/topics/topic.rb:29:in `method_missing'
/openc3/lib/openc3/microservices/decom_microservice.rb:53:in `block in run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/openc3/lib/openc3/microservices/decom_microservice.rb:52:in `run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/microservices/microservice.rb:55:in `run'
multi_microservice.rb:43:in `block (2 levels) in run'
```

ryanmelt, 2023-12-21:
You can't use openc3/script inside of internal code, because it goes through the external API and requires authentication.  This only used to work due to more credentials than necessary being available in the system.

For internal trusted code you can use the api directly.

require 'openc3/api/api' and then "include Api" in your class.

scandey, 2023-12-21:
Thank you, that fixed it!

closed 2023-12-21 by scandey