OpenC3/cosmos #1003: Automatically upload and install PLUGIN

omermesser, 2023-12-24:
Is there a way or an API to upload a plugin ".gem" file and install it automatically?
also deleting the previous plugin.
[image]

ryanmelt, 2023-12-24:
Yes you can do this with our openc3cli command line utility

```
export OPENC3_API_USER=admin
export OPENC3_API_PASSWORD=admin
export OPENC3_API_HOSTNAME=localhost
export OPENC3_API_PORT=2900
export OPENC3_KEYCLOAK_URL=http://localhost:2900/auth

openc3cli load openc3-cosmos-demo-1.0.0.gem
```

Unload is also supported, but you need the full name including the __timestamp part

closed 2023-12-24 by ryanmelt

omermesser, 2023-12-25:
I have added the ENVs the same way you configured.
tried executing the load command.
Yet I receive the following error:
```
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:99:in `rescue in plugin_install_phase1': plugin_install_phase1 failed due to ConnectionFailed : Failed to open TCP connection to localhost:2900 (Address not available - connect(2) for "::1" port 2900) (RuntimeError)
```

this is the full call stack:
```
..\..\..\..\openc3 cli load openc3-cosmos-block-one-targets-5.10.0.gem
Error response from daemon: network with name openc3-cosmos-network already exists
ruby: warning: shebang line ending with \r may cause problems
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:99:in `rescue in plugin_install_phase1': plugin_install_phase1 failed due to ConnectionFailed : Failed to open TCP connection to localhost:2900 (Address not available - connect(2) for "::1" port 2900) (RuntimeError)
/usr/lib/ruby/3.2.0/resolv-replace.rb:25:in `initialize'
/usr/lib/ruby/3.2.0/resolv-replace.rb:25:in `initialize'
/usr/lib/ruby/3.2.0/net/http.rb:1271:in `open'
/usr/lib/ruby/3.2.0/net/http.rb:1271:in `block in connect'
/usr/lib/ruby/gems/3.2.0/gems/timeout-0.4.1/lib/timeout.rb:186:in `block in timeout'
/usr/lib/ruby/gems/3.2.0/gems/timeout-0.4.1/lib/timeout.rb:193:in `timeout'
/usr/lib/ruby/3.2.0/net/http.rb:1269:in `connect'
/usr/lib/ruby/3.2.0/net/http.rb:1248:in `do_start'
/usr/lib/ruby/3.2.0/net/http.rb:1237:in `start'
/usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:112:in `request_with_wrapped_block'/usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:102:in `perform_request'
/usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:66:in `block in call'
/usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/adapter.rb:45:in `connection'
/usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:65:in `call'
/usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/request/url_encoded.rb:25:in `call'
/usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/rack_builder.rb:153:in `build_response'
/usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/connection.rb:444:in `run_request'
/usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/connection.rb:280:in `post'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:152:in `_make_request'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:117:in `_make_token'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:82:in `block in token'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:78:in `synchronize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:78:in `token'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:89:in `block in plugin_install_phase1'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:80:in `open'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:80:in `plugin_install_phase1'
/openc3/bin/openc3cli:479:in `load_plugin'
/openc3/bin/openc3cli:635:in `<main>'
Response:

        from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:71:in `plugin_install_phase1'
        from /openc3/bin/openc3cli:479:in `load_plugin'
        from /openc3/bin/openc3cli:635:in `<main>'
/usr/lib/ruby/3.2.0/resolv-replace.rb:25:in `initialize': Failed to open TCP connection to localhost:2900 (Address not available - connect(2) for "::1" port 2900) (Faraday::ConnectionFailed)
        from /usr/lib/ruby/3.2.0/resolv-replace.rb:25:in `initialize'
        from /usr/lib/ruby/3.2.0/net/http.rb:1271:in `open'
        from /usr/lib/ruby/3.2.0/net/http.rb:1271:in `block in connect'
        from /usr/lib/ruby/gems/3.2.0/gems/timeout-0.4.1/lib/timeout.rb:186:in `block in timeout'
        from /usr/lib/ruby/gems/3.2.0/gems/timeout-0.4.1/lib/timeout.rb:193:in `timeout'
        from /usr/lib/ruby/3.2.0/net/http.rb:1269:in `connect'
        from /usr/lib/ruby/3.2.0/net/http.rb:1248:in `do_start'
        from /usr/lib/ruby/3.2.0/net/http.rb:1237:in `start'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:112:in `request_with_wrapped_block'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:102:in `perform_request'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:66:in `block in call'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/adapter.rb:45:in `connection'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-net_http-3.0.2/lib/faraday/adapter/net_http.rb:65:in `call'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/request/url_encoded.rb:25:in `call'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/rack_builder.rb:153:in `build_response'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/connection.rb:444:in `run_request'
        from /usr/lib/ruby/gems/3.2.0/gems/faraday-2.8.1/lib/faraday/connection.rb:280:in `post'
        from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:152:in `_make_request'
[... cut ...]
        from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/utilities/authentication.rb:78:in `token'
        from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:89:in `block in plugin_install_phase1'
        from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:80:in `open'
        from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.10.0/lib/openc3/script/plugins.rb:80:in `plugin_install_phase1'
        from /openc3/bin/openc3cli:479:in `load_plugin'
        from /openc3/bin/openc3cli:635:in `<main>'
```

ryanmelt, 2023-12-25:
The directions are for using a host ruby installation.  (openc3cli not openc3.sh cli).
Local ruby installation with the openc3 gem installed.

omermesser, 2024-01-01:
I do not understand how to run it if not from the openc3.bat cli.
do you mind briefly explain how to do it?

omermesser, 2024-01-01:
I have tried installing the openc3 gem file. but with no success.
the only place opnec3cli install is mentioned is here: [https://docs.openc3.com/docs/guides/bridges#host-requirements-for-running-bridges](url)
Is there a documentation saying how to make it work? or can you provide an explanation.
maybe I am missing something regarding the install.
console log: [console log.txt](https://github.com/OpenC3/cosmos/files/13804663/console.log.txt)

mkmf log: [mkmf.log](https://github.com/OpenC3/cosmos/files/13804658/mkmf.log)

omermesser, 2024-01-01:
I failed to load the PLUGIN.
I managed to install the openc3cli gem, 
I set the ENV with the correct values as shown in the above comment [https://github.com/OpenC3/cosmos/issues/1003#issuecomment-1868546537](url)

But I receive an error shown in the console log:
[console.log](https://github.com/OpenC3/cosmos/files/13804890/console.log)

Please advise

ryanmelt, 2024-01-04:
I updated my earlier comment.  If you are using open source edition.  Do not define a OPENC3_API_USER.

```
export OPENC3_API_USER=admin # Only For Enterprise Edition
export OPENC3_API_PASSWORD=admin
export OPENC3_API_HOSTNAME=localhost
export OPENC3_API_PORT=2900
export OPENC3_KEYCLOAK_URL=http://localhost:2900/auth # Only for Enterprise Edition
```

omermesser, 2024-01-29:
> I updated my earlier comment. If you are using open source edition. Do not define a OPENC3_API_USER.
>
> ```
> export OPENC3_API_USER=admin # Only For Enterprise Edition
> export OPENC3_API_PASSWORD=admin [...]

I have managed to solve it using this solution, thanks!