OpenC3/cosmos #1008: Help with python interfaces

bradenvirdellgenevatech, 2023-12-29:
Hello,

I have been trying install a gem with some of the python interfaces. So far I have been unable to get the interface to show up in the CmdTlmServer tool under the interfaces section. 

To get to this point I have done the following
1. Started up a Cosmos Server using 5.13.0 
2. Ensured the containers I am building my gems off of are also 5.13.0 
3. I generated a plugin the normal way
4. I generated a target like so `./openc3.sh cli generate target <NAME> --python` (replaced the NAME variable)
5. Made the plugin look like below
[image]

6. Built the gem like so `./openc3.sh cli rake build VERSION=<X.X.X>` (replaced the version variables)

Then uploaded the gem.
I see the interface under the admin console but i do not see it under the CmdTlmServer.

At this point I don't know what I did wrong or what step I missed. The demo which has a python based interface works properly. So I dont think its with the server but with my building of the gem somewhere.

Any help is appreciated.
Thanks,
Braden

ryanmelt, 2023-12-29:
I just tried this with our generator and it is working for me.

Most likely the interface microservice is crashing for some reason for you.   That should show up in the operator container logs:

docker logs <hash of operator container>

ryanmelt, 2024-01-03:
@bradenvirdellgenevatech Any update on this?

bradenvirdellgenevatech, 2024-01-04:
After reading the logs its cause I was sending a `nil ` instead of a `None` for the read timeout. I should have known to change that it just didnt occur to me so my bad.  I think it would good to have some examples for the python interfaces in the documentation like you do for the ruby ones to clarify how to use them.

A second question off this would be how do custom python interfaces. I assume that when the interface is loaded it will look in the gem's lib directory and pull which ever one matches it just like it does for ruby?

ryanmelt, 2024-01-07:
Finding the python files in the lib folders like Ruby is definitely the goal.   We haven't built many python plugins yet though, so let us know if it doesn't work like you expect.

closed 2024-01-07 by ryanmelt