OpenC3/cosmos #1010: Help w/ setting up plugin and tlm file for LENGTH protocol

rogueWookie, 2024-01-03:
Hi ~ 

I've been at it over the last couple of days trying to get a simple LENGTH protocol working over a serial connection between cosmos and my raspberrypi. I've tried so many things but I keep getting the same error every time ... 

```
WARN	DEFAULT__INTERFACE__RPI_INT   RPI_INT UNKNOWN packet length: 14 starting with: EFBEADDE06000000020000000810

WARN	DEFAULT__INTERFACE__RPI_INT   RPI_INT UNKNOWN packet length: 14 starting with: EFBEADDE0600000002000000070E
```

I've tried changing so many aspects of the `plugin.txt` file that I'm completely lost now on what else to do. Would you be so kind to give my configs a second look please.

```bash
cosmos-project/openc3-cosmos-rpi on  main [!?] took 14s 
❯ cat plugin.txt 
# Set VARIABLEs here to allow variation in your plugin
VARIABLE rpi_target_name RPI
TARGET RPI <%= rpi_target_name %>

INTERFACE <%= rpi_target_name %>_INT serial_interface.rb /dev/ttyUSB0 /dev/ttyUSB0 115200 NONE 1 nil 10 LENGTH 32 32 8 1 BIG_ENDIAN 0 0xDEADBEEF
    MAP_TARGET <%= rpi_target_name %>
``` 

```bash
cosmos-project/openc3-cosmos-rpi
❯ cat targets/RPI/cmd_tlm/tlm.txt 
TELEMETRY RPI CNT_INFO BIG_ENDIAN "Example Counters"
    #   Keyword     Name        BitSize     Type        ID      Description
    APPEND_ITEM     syn_mark    32          UINT                "Sync Marker"
    APPEND_ITEM     len_pack    32          UINT                "Length of the Packet"
    APPEND_ID_ITEM  pkt_apid    32          UINT        2       "Packet Application Identifier"
    APPEND_ITEM     x           8           UINT                "Increments by 1"
    APPEND_ITEM     y           8           UINT                "Increments by 2"
```

ryanmelt, 2024-01-03:
Your data appears to be LITTLE_ENDIAN.   Replace BIG_ENDIAN with LITTLE_ENDIAN.

rogueWookie, 2024-01-03:
So embarrassing, that was the issue. Thanks Ryan, really appreciate the help.

rogueWookie, 2024-01-03:
Hi @ryanmelt 

The serial interface I have defined (with the configs above) seems to be working except every so often it's like it gets out of sync. At first I thought it was on my target side. However, I've defined a static byte stream (so nothing is changing) on my target and send the byte stream every second but it still gets out of sync every so often. Have you encountered something like this? Here is what the out of sync tends to look like. 

```txt
2024-01-03 15:30:08.028	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 9 starting with: 000000020000000101
2024-01-03 15:30:08.012	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 5 starting with: EFBEADDE06
2024-01-03 15:30:07.517	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 3 starting with: 000101
2024-01-03 15:30:07.501	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 11 starting with: EFBEADDE06000000020000
2024-01-03 15:29:49.596	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 13 starting with: BEADDE06000000020000000101
2024-01-03 15:29:49.580	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 1 starting with: EF
2024-01-03 15:29:49.084	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 8 starting with: 0000020000000101
2024-01-03 15:29:49.068	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 6 starting with: EFBEADDE0600
2024-01-03 15:29:48.572	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 4 starting with: 00000101
2024-01-03 15:29:48.556	WARN	DEFAULT__INTERFACE__RPI_INT RPI_INT UNKNOWN packet length: 10 starting with: EFBEADDE060000000200
```

ryanmelt, 2024-01-04:
I'm confused because what you are seeing shouldn't be possible with a sync pattern defined.   What is your current INTERFACE config?

ryanmelt, 2024-01-04:
There has to be some kind of bug with our sync pattern logic.   Sync patterns are always "effectively" BIG ENDIAN.  The bytes need to be specified in the order received.  So your sync pattern should be specified as 0xEFBEADDE not 0xDEADBEEF.

But it should not be working at all with how you have it specified, so I need to dig in.

ryanmelt, 2024-01-04:
Are you sure you were using the LENGTH protocol?  This feels like the BURST protocol with no arguments.

rogueWookie, 2024-01-06:
Hi @ryanmelt 

I don't have my work laptop I was doing the testing on but today I did some tinkering of my own here at home. Everything appears to be working fine now. I've [__uploaded my cosmos plugin content__](https://github.com/rogueWookie/openc3-cosmos-rpi) and my [__python app on the rpi__](https://github.com/rogueWookie/openc3-cosmos-rpi/tree/master/targets/RPI/target_apps) if you want to see what I've ended up with.

I'm afraid I don't have a good answer for what happened. My only suspicion at this point is my workflow (when I go about editing cosmos files e.g. `plugin.txt`, `tlm.txt`, `compose.yaml`) might be causing some issues. Maybe you can advise here a bit below on what my workflow is now but other than that everything is working nicely. Thanks again for the help!

__WORKFLOW__

- Upon any editing of cosmos files (e.g. `plugin.txt`, `tlm.txt`, `compose.yaml`) I am not stopping the containers. I leave them running because in your documentation I don't see any indication I have to do that (e.g. `openc3.sh stop`).
- Following any edits to any file in my plugin directory I just build a new gem via `rake` just as your documentation points out. After it indicates a successful build I upload/install the new version (just as the docs say). 

Maybe I should be restarting the containers (e.g. `openc3.sh stop/start`). It just wasn't pointed out in the documentation that was required. However, I presume if I modify `compose.yaml` or `.env` I would need to restart the containers in that case.

ryanmelt, 2024-01-06:
Your workflow sounds fine.   As you said, you should only have to restart containers if you change the compose or .env files.
My guess is that you were running with an old plugin.txt after upgrading.   When you upgrade it should show you a diff dialog.  You have to merge the changes otherwise it will keep running with your old config.

closed 2024-01-06 by ryanmelt

rogueWookie, 2024-01-06:
That was exactly it Ryan. I wasn't merging changes in plugin.txt which meant I was running with BURST. 

In that pop up window showing the diff I can barely see the arrow button to merge. It's really small so it just wasn't obvious to me and so I was clicking install vs doing the merge first. Thanks again!