OpenC3/cosmos #1011: Plugins Permission Denied

JTrulear93, 2024-01-03:
I have been trying to get my plugins uploaded to my server, but every time I attempt to do so, I get an HTTP 500 Permission denied error. Has this happened to anyone before, and does anyone know how to get around this so I can test my plugins? Any help would be greatly appreciated. Thank you.

[image]

JTrulear93, 2024-01-03:
From what I can tell, I am running OpenC3 COSMOS Open Source 5.11.1. At least, that what it says on the footer of the webpage.

ryanmelt, 2024-01-03:
If you have a support contract, reach out to support@openc3.com and we can help.

The errors you are getting are permission errors writing to the /gems volume mount.   That generally would happen if the user your container is running as is different than the user who owns the /gems volume.

Are you sharing a cosmos install on a linux system and trying to run it with different user accounts?

JTrulear93, 2024-01-03:
The cosmos install is on a linux server, yes. But we all use one shared account when trying to run it. However, I built my gem files on my local Windows machine because I cannot generate them on the linux server. Will gem files built on different machines not run correctly?

I can reach out to the support team if that's preferred.

ryanmelt, 2024-01-03:
Building the gem on windows should be fine.

You need to docker exec into the openc3-cosmos-cmd-tlm-api container to figure out what permissions are unhappy

```
docker ps
docker exec -it <hash of openc3-cosmos-cmd-tlm-api container> sh
ls -la /gems
whoami
```

JTrulear93, 2024-01-03:
I did the steps you provided, and this is what came up. I have no idea what it means though.

[image]

ryanmelt, 2024-01-03:
Your 500 errors are because the container is running as user_id 1268369 and the files are owned by user_id 833154

Those are some crazy user_ids though.  Your IT organization must have some fancy setuid settings configured.

You can try this to see if it will fix it (note exec is execing in as root this time):
```
docker ps
docker exec -it --user root <hash of openc3-cosmos-cmd-tlm-api container> sh
chown -R 1268369 /gems
```

JTrulear93, 2024-01-03:
That seemed to fix it! I was able to upload my plugin (with errors, but that's still an improvement!). Thank you very much.

closed 2024-01-03 by ryanmelt