OpenC3/cosmos #1023: Cosmos 5.12 Python Target cannot send 0 bitsize telemetry strings

josephmartin09, 2024-01-08:
**Describe the bug**
My python target cannot send telemetry strings with bitsize 0 (which I assume means variable length) without crashing.  Changing the bitsize to a fixed number such as 2048 seems to fix the issue

**To Reproduce**
Create a derived class of SimulatedTarget, such as linked below. It seems writing the "label" field stops me from seeing telemetry via the interface
```python
import time
import base64
import random
import threading
from io import BytesIO
from openc3.utilities.simulated_target import SimulatedTarget
from openc3.packets.structure import Structure
from openc3.packets.packet import Packet
from openc3.system.system import System
from openc3.top_level import kill_thread

class OdtnSender(SimulatedTarget):

    def __init__(self, target_name):
        super().__init__(target_name)
        self.target = System.targets[target_name]

    def set_rates(self):
        self.set_rate("STATUS", 100)

    def tick_period_seconds(self):
        return 0.1

    def tick_increment(self):
        return 10

    def write(self, packet):
        pass

    def read(self, count_100hz, time):
        pending_packets = self.get_pending_packets(count_100hz)

        for packet in pending_packets:
            match packet.packet_name:
                case "STATUS":
                    new_val = random.randint(1, 10)
                    packet.write("value", new_val)
                    packet.write("bool",  new_val >= 5)
                    packet.write("label", f"This is a label: {new_val}")

        return pending_packets

```

**Expected behavior**
Based on the ruby examples, I would expect a 0 bitsize string to dynamically change based on string length.  Perhaps this is a misunderstanding on my part?

**Screenshots**
**Does not produce telemetry**
[image]

**Works correctly**
[image]

**Environment (please complete the following information):**

- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.12
- Browser: Firefox

ryanmelt, 2024-01-10:
You don't appear to be doing anything wrong.  It's probably a bug in the new Python code, we'll look into it.

jmthomas, 2024-01-11:
We currently only allow writing bytes so you can encode your string like so:
```
packet.write("label", f"This is a label: {new_val}".encode(encoding="utf-8"))
```
I'll update this going forward to allow regular strings.

closed 2024-01-13 by jmthomas