OpenC3/cosmos #1024: Is there a way to access minio on 5.13.0?

GTI-Dakota, 2024-01-08:
Hello,

I am upgrading from 5.8.x to 5.13.0 and checking out difference between versions. It looks like I'm not able to access minio via the /minio subdirectory anymore. Is this due to the minio change listed in the release notes? Minio has been especially helpful for downloading entire log folders and removing multiple objects easily - which the bucket explorer does not seem to support.

Thanks,

Dakota

ryanmelt, 2024-01-08:
Newer versions of minio console don't like working through the Traefik proxy, so it has to be exposed manually.  
Edit your compose.yaml to be like this:

Changes:
1. Add ports section to expose port 9001
2. Comment out the MINIO_BROWSER_REDIRECT_URL environment variable

You can then access through localhost:9001

```
services:
  openc3-minio:
    user: "${OPENC3_USER_ID:-1001}:${OPENC3_GROUP_ID:-1001}"
    image: "${OPENC3_ENTERPRISE_REGISTRY}/${OPENC3_ENTERPRISE_NAMESPACE}/openc3-enterprise-minio${OPENC3_IMAGE_SUFFIX}:${OPENC3_ENTERPRISE_TAG}"
    ports:
      - "127.0.0.1:9001:9001"
    volumes:
      - "openc3-bucket-v:/data"
      - "./cacert.pem:/devel/cacert.pem:z"
    command: server --address ":9000" --console-address ":9001" /data
    restart: "unless-stopped"
    logging:
      driver: "json-file"
      options:
        max-size: "10m"
        max-file: "3"
    environment:
      MINIO_ROOT_USER: "${OPENC3_BUCKET_USERNAME}"
      MINIO_ROOT_PASSWORD: "${OPENC3_BUCKET_PASSWORD}"
      # Domain doesn't really matter but it's required. We really want the /minio path.
      # This is handled by our traefik configuration via rule: PathPrefix(`/minio`)
      # and forwarded on to the console at http://openc3-minio:9001
      # MINIO_BROWSER_REDIRECT_URL: "http://openc3.com/minio"
      SSL_CERT_FILE: "/devel/cacert.pem"
      CURL_CA_BUNDLE: "/devel/cacert.pem"
      REQUESTS_CA_BUNDLE: "/devel/cacert.pem"
      NODE_EXTRA_CA_CERTS: "/devel/cacert.pem"
```

GTI-Dakota, 2024-01-08:
Awesome, thanks!

closed 2024-01-08 by GTI-Dakota