OpenC3/cosmos #1025: COSMOS will pull an old protocol file if the gem is deleted (and packages removed on the /tools/admin/packages subdirectory)

GTI-Dakota, 2024-01-08:
If I make a change to a protocol and then upgrade (or delete and re-upload) a gem, the protocol will not change if it shares the same name w/ a previously uploaded file within that gem. In older COSMOS versions, the workaround I found was to delete the gem from the admin panel, and then remove it from the then-named "gems" tab. Now, this method is not working.

For ease of version validation, I tossed in a log message that prints out the protocol version in initialize, so I know which protocol is being referenced. As of 5.13.0, I cannot get the version number to increment across uploads.
[image]

**To Reproduce**
1. Upload a gem with a protocol in the [gem_name]/lib directory.
2. Change the protocol file, rebuild gem
3. Upgrade or delete+re-upload gem
4. Observe the cmd/tlm log

**Expected behavior**
An old protocol file will be used.

**Screenshots**
My version is 16
[image]

Version printed out is 15
[image]

**Environment (please complete the following information):**
- OS: Linux Ubuntu 20.04.4 LTS
- OpenC3 COSMOS Version 5.13.0
- Browser: N/A

ryanmelt, 2024-01-08:
Going to probably need to debug this one with you Dakota.  If the gem is competely deleted everywhere it shouldn't be possible to be still being used.

GTI-Dakota, 2024-01-08:
Okay, I think I found the issue. The protocol still existed (with the same name) as another independent gem that was deleted from the admin console but still in the "packages" list. Once deleted, I was able to reupload my gem and the protocol version updated. For some reason, gem DEF was pulling the protocol file from gem ABC. Does COSMOS pull the first file it finds with the name or does it check the newly installed gem's folder first?

GTI-Dakota, 2024-01-08:
Makes sense. I'll be careful regarding duplicate protocol names in the future 👍

closed 2024-01-08 by ryanmelt