OpenC3/cosmos #1049: LITTLE_ENDIAN PARAMETER in BIG_ENDIAN COMMAND Not Working

tsjordan-eng, 2024-01-13:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
When a particular PARAMETER of a COMMAND packet is set to LITTLE_ENDIAN, the resulting bytes are **not** swapped as they should be with this setting.

**To Reproduce**
Steps to reproduce the behavior:

1. Create a BIG_ENDIAN COMMAND by appending `BIG_ENDIAN` after the COMMAND description
2. Change one of the PARAMETERS (greater than 1 byte) to LITTLE_ENDIAN by appending `LITTLE_ENDIAN` after the description
3. Send the command
4. Receive the command with your target, notice that the value is not little-endian

**Expected behavior**
Though all the parameters in the command should default to big-endian, the particular field should be little-endian

**Environment (please complete the following information):**

- OS: RHEL 9, Podman
- OpenC3 COSMOS Version 5.12.0
- Browser: Edge

jmthomas, 2024-01-13:
I am unable to reproduce. I created a test case in the linked PR as well as changed the demo SETPARAMS command to have LITTLE_ENDIAN parameters. I sent a command with PARAM1 set to 1, PARAM2 to 2, etc and when I uncommented the buffer line in the sim_inst.rb file I get the following:

```
SETPARAMS packet: 00000000: 13 E7 C0 00 00 0C 00 04 01 00 02 00 03 00 04 00                  
00000010: 0A 00 
```

You'll notice the `00 04` is the PKTID field and then parameters start. The bytes are swapped: `01 00`, `02 00`, etc.

tsjordan-eng, 2024-01-15:
Thanks for your quick response! We'll double-check our setup, see what you did, and see if we're mistaken or have found an exception.

jmthomas, 2024-01-17:
@tsjordan-eng we're going to cut a new release this week so please let me know if you find anything. I want to make sure this is resolved for the release.

tsjordan-eng, 2024-01-18:
Not sure why it's working now, but it is. This may be related to #940 where the plugin said it updated or uninstalled or downgraded, but it really didn't. I'm trying to update to 5.13 but having issues so we'll see. For now it looks like the ENDIANness is working.

PR #1051 (2024-01-18): Test command endianness
closes #1049