OpenC3/cosmos #1059: Lost data/accessing redis

bradenvirdellgenevatech, 2024-01-17:
Hello,

For context:

Yesterday
We had a script running that was using the StreamingWebSocket Api to collect data. At the same time one of our engineers ran the data extractor tool over the same time period for 1 packet. Both were ran over the same ~10 min period.

Once both had completed, 
The script only provided 1 or 2 received packets per packet definition. We know that was more data than that as we were watching it as it was coming in real time..
the data extractor tool got all the data we expected from that packet definition.
 
Today
I have tried running a manual data extract over the same time, and now Cosmos is stating that there is no data there at all.

Questions:
Is it possible that we some how lost data due to the extractions happening over the same time?

Is there an easy way to access redis? (used to be able to connect to it through a port) Seems like helm can do it but I am unsure.

COSMOS Version 5.9.0
System: Linux Ubuntu 22.04.2 LTS

Any help is appreciated

Thanks,
Braden

ryanmelt, 2024-01-17:
Can you look in BucketExplorer and confirm the decom files are there?

Bucket: logs
/DEFAULT/decom_logs/tlm/TARGET_NAME/DATE/

bradenvirdellgenevatech, 2024-01-17:
I see decom long at 2024-01-17 01:20:17 UTC which is about 10 mins after the issue seems to have occurred.

ryanmelt, 2024-01-17:
So does it look like something is missing?

bradenvirdellgenevatech, 2024-01-17:
From what I can tell it seems like it has data starting from the end of the pervious decom file onward as it should. So I don't think there is a missing file.

ryanmelt, 2024-01-17:
Ok.  When you stream over that time frame, you get nothing though?
If you look at the logs from openc3-cosmos-cmd-tlm-api, do you see any errors after you try to stream?

ryanmelt, 2024-01-17:
Can you try on 5.13.0?  I think you may be running into an issue we already fixed with streaming time ranges.

bradenvirdellgenevatech, 2024-01-17:
This being a production server it would be quite difficult to try that, I am not seeing any erros ins the cmd_tlm_api

ryanmelt, 2024-01-17:
Ok.  If you can download the necessary files from BucketExplorer, I can provide you a Ruby script that will be able to read the data out of the file directly.  You will need a local ruby installation with the openc3 gem installed.

bradenvirdellgenevatech, 2024-01-17:
Sounds good, I should have all that available already

ryanmelt, 2024-01-18:
Here's a simple script to read the file.   Note that the file must be ungziped first.

# print_packet_log.rb
# Usage: ruby print_packet_log.rb FILENAME.bin

```
require 'openc3'

plr = OpenC3::PacketLogReader.new
plr.each(ARGV[0], false) do |packet|
  if OpenC3::Packet === packet
    puts packet.buffer(false).formatted
  else
    pp packet.read_all(:WITH_UNITS)
  end
  puts
end
```

bradenvirdellgenevatech, 2024-01-18:
Works perfectly, as of now I think we got all the missing data (hard to tell until I parse it out to some CSV's). Thank you very much for the assistance

closed 2024-01-19 by ryanmelt