OpenC3/cosmos #1068: Unable to use Python "pillow"

timothyscherer, 2024-01-23:
I am trying to use `matplotlib` in a script. I have added it to my `requirements.txt` along with `numpy`, which installs the following packages (list from admin/packages):

```
contourpy-1.2.0
cycler-0.12.1
fonttools-4.47.2
kiwisolver-1.4.5
matplotlib-3.8.2
numpy-1.26.3
pillow-10.2.0
```

I have no problems importing and using `numpy`, but get an error trying to import from `matplotlib`:

```py
import numpy
from matplotlib import pyplot as plt
```
```
... "/gems/python_packages/lib/python3.11/site-packages/matplotlib/colors.py\", line 52, in <module>
from PIL import Image
ImportError: cannot import name 'Image' from 'PIL' (unknown location)\n", "type": "log"}
```

PIL is the namespace for the `pillow` package. Naturally I tried to import PIL directly and print the handle, and I did the same with numpy to compare the results.

```py
import numpy
print(numpy)
import PIL
print(PIL)
```
```
<module 'numpy' from '/gems/python_packages/lib/python3.11/site-packages/numpy/__init__.py'>
<module 'PIL' (<openc3.utilities.target_file_importer.MyLoader object at 0x7f8d689b6f90>)>
```

It appears that the library is not being loaded properly, hence why it is not found when I attempt to import it. I was able to run this Python script from inside OpenC3's container without issue, using the same Python install and packages.

- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.14.0
- Browser: Firefox

ryanmelt, 2024-01-24:
I think this isn't working because PIL is all caps and our mods to make it import files from a bucket look for a module name that is all caps (like our target names).   We'll have to see if we can make that smarter.

jmthomas, 2024-02-07:
closed by #1082

closed 2024-02-07 by jmthomas