OpenC3/cosmos #1074: Unable to import python files

Lakshmi5785-dev, 2024-02-01:
Hi Team,
    I am using opensource license where the scope is DEFAULT.
I installed v5.13.0.
I am trying to run my own python scripts but unable to import the files.
For example I have 2 files in the cosmos\plugins\DEFAULT\targets_modified folder (test1.py and test2.py) When I try to import test1 in the test2.py (from targets_modified.test1 import *) , I am getting module not found error.
This works fine incase of ruby files.
Please help.

jmthomas, 2024-02-02:
If you want the code instrumented (line by line execution) you need to do the following (assuming your target is called DEFAULT):
```
load_utility("DEFAULT/procedures/test1.py")
```
If you want to simply load it try this:
```
from DEFAULT.procedures.test1 import *
```

Lakshmi5785-dev, 2024-02-02:
Hey Thank you for such a quick response. I got it from your documentation
after I raised the ticket.
And able to solve the issue.
Once again thank you so much :)

closed 2024-02-07 by jmthomas