OpenC3/cosmos #1094: Unable to run OpenC3 in devepment mode

Lakshmi5785-dev, 2024-02-12:
Hello Team,
   I am trying to run the OpenC3 in the development mode as suggested in the below link.
https://docs.openc3.com/docs/development/developing(Running a Backend Server)
When I try to run the command bundle exec rails s I am facing the below issues please help .

[image]

[image]

Please help

jmthomas, 2024-02-13:
First I see you're running Ruby 3.3. This _may_ work but we have not yet tested it. However, this is not your problem. We forgot a few steps before issuing `bundle exec rails s`. First you need to source the .env file, then set a few variables to localhost:

```
openc3-cosmos-cmd-tlm-api % set -a; source ../.env; set +a
openc3-cosmos-cmd-tlm-api % export OPENC3_REDIS_HOSTNAME=localhost
openc3-cosmos-cmd-tlm-api % export OPENC3_REDIS_EPHEMERAL_HOSTNAME=localhost
```
Now `bundle exec rails s` should work. I will update the documentation.

Lakshmi5785-dev, 2024-02-13:
May I know which version of ruby are you referring to?

Lakshmi5785-dev, 2024-02-13:
> I can see in your stack trace C:/Ruby33-x64. Ruby 3.3 was released on Dec
> 25, 2023 and has not yet been incorporated into Alpine which we base all
> our containers on. You should install Ruby 3.2 for now.

I have tried with 3.2.3 , here bundle install itself is failing :(
[image: image.png]

Lakshmi5785-dev, 2024-02-13:
By the way I am running it on windows machine.

jmthomas, 2024-02-13:
Actually in Windows you should simply run `dev_server.bat`

Lakshmi5785-dev, 2024-02-14:
Now I am running dev_server.bat

With Ruby 3.3
[image: image.png]

with ruby 3.2
[image: image.png]

ryanmelt, 2024-02-14:
Your images are not showing up

Lakshmi5785-dev, 2024-02-14:
Hey Ryan Good evening :).
Here I am attaching the images.

ryanmelt, 2024-02-14:
You cannot reply to github emails with images.   You need to add them on the website.

Lakshmi5785-dev, 2024-02-14:
Sure.
I have updated the ticket/comment. Please check
Unable to run cosmos in development mode · Issue #15 ·
OpenC3/cosmos-project (github.com)
<https://github.com/OpenC3/cosmos-project/issues/15>

closed 2024-02-14 by ryanmelt