OpenC3/cosmos #1097: Can templated commands/telemetry differentiate between items in a response without a delimiter between them?

mkalin2, 2024-02-13:
I have a template command defined as seen below:
```
COMMAND ENCODER GET_ID BIG_ENDIAN "Retrieve encoder serial number and part number"
    APPEND_ID_PARAMETER CMD_ID 24 STRING "$0V"
    APPEND_PARAMETER CMD_TEMPLATE 512 STRING "<CMD_ID>"
    APPEND_PARAMETER RSP_TEMPLATE 512 STRING "<TELEM_ID><PART_NUM>,<SERIAL_NUM>"
    APPEND_PARAMETER RSP_PACKET 512 STRING "ID"
```
As you can see in the RSP_TEMPLATE definition, there are two separate telemetry items that are returned by my device without a delimiter between them (TELEM_ID and PART_NUM).

Here is how my telemetry response packet is defined:
```
TELEMETRY ENCODER ID BIG_ENDIAN "Encoder serial number and part number"
    APPEND_ID_ITEM TELEM_ID 24 STRING "*0V"
    APPEND_ITEM PART_NUM 104 STRING "Encoder part number"
    APPEND_ITEM SERIAL_NUM 64 STRING "Encoder serial number"
```
This is what the device response actually looks like on the serial line: `*0V924-60013-001,YY109477`. "*0V" is the telemetry packet ID and "924-60013-001" is the part number.

But this is what the raw packet looks like from inside of COSMOS: 
```
Address   Data                                             Ascii
---------------------------------------------------------------------------
00000000: 2a 30 56 00 00 00 00 00 00 00 00 00 00 00 00 00  *0V             
00000010: 59 59 31 30 39 34 37 37                           YY109477        
```
When COSMOS receives this response packet, I get the following error: `ENCODER_INT: Could not write value *0V924-60013-001 due to value of 16 bytes does not fit into 3 bytes for data_type STRING`

It is reading the first 16 bytes of the telemetry response as the TELEM_ID and trying to cram them into the 3 bytes allocated in the packet for TELEM_ID, then when that fails it throws out the remaining 13 bytes (the part number) and raises the above error. It then leave the PART_NUM empty and moves onto the SERIAL_NUM.

Instead, I would like COSMOS to split those 16 bytes into a 3 byte telemetry ID and a 13 byte part number. Is there a way to accomplish this without some kind of delimiter in the device response?

Here is what the telemetry looks like in the packet viewer:
[image]
(normally I ignore the TELEM_ID but it is included here for debugging)

ryanmelt, 2024-02-13:
Without a delimiter you will either need to do something custom, or read into a single field, and then break apart into DERIVED items that have conversions that know to take the first three characters etc.

mkalin2, 2024-02-14:
I don't really need those first three bytes except as an ID. If I set the protocol to discard those three bytes can I still use them as an ID?

mkalin2, 2024-02-14:
I am trying your suggestion of making DERIVED telemetry items, and I get the following error when trying to use a DERIVED item as my telemetry ID: 
```
Error: DERIVED data type not allowed for Identifier
Usage: APPEND_ID_ITEM <ITEM NAME> <BIT SIZE> <TYPE: INT/UINT/FLOAT/STRING/BLOCK/DERIVED> <ID VALUE> <DESCRIPTION (Optional)> <ENDIANNESS (Optional)>
```
In the documentation and the above error message it says I can set the type to be DERIVED. Is this a bug?

jmthomas, 2024-02-15:
ID_ITEMs can not be DERIVED. ID_ITEMs are used to identify the packet so it doesn't really make sense to use a DERIVED item for this. This is a documentation error.

closed 2024-06-18 by mkalin2