OpenC3/cosmos #1099: Errors running 5.14.2 with Podman

mdpalme, 2024-02-14:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
5.14.2 openc3.sh run fails to launch a working session when using Podman:
[start_errors.txt](https://github.com/OpenC3/cosmos/files/14282233/start_errors.txt)

**To Reproduce**
Steps to reproduce the behavior:

1. `git clone https://github.com/openc3/cosmos-project.git cosmos-myproject`
2. `cd cosmos-myproject`
3. Run Linux: `./openc3.sh run`
4. `docker-compose logs -f | grep -i "error"` (See attached file for output)
5. No server is running when trying to connect a web browser to http://localhost:2900/
6. I've tried the above steps again after stopping all containers and running `podman volume prune` and `podman system prune --all` 
7. I've tried the above steps again after running the `migrate_5_12_to_5_13.sh`
 
**Expected behavior**
I expect the server to be running when trying to connect with a web browser.
I am able to get 5.14.2 running with Docker/Windows.... just not Podman/RHEL9.

**Environment (please complete the following information):**

- OS: RHEL 9.1
- OpenC3 COSMOS Version 5.14.2
- Podman 4.6.1
- Docker Compose v2.16.0
- Browser: Chrome

[start_errors.txt](https://github.com/OpenC3/cosmos/files/14282303/start_errors.txt)

ryanmelt, 2024-02-14:
I'm pretty sure podman is bugged on redhat 9.1.  Please try on 9.2

ryanmelt, 2024-02-14:
You errors file looks like it is at least partially running though.   What does "podman ps" show?   The init container might be restarting for some reason

mdpalme, 2024-02-14:
`posman ps -a`:
[image]

ryanmelt, 2024-02-14:
Need to see that after more than 10 seconds.  Give it a couple minutes. :)

ryanmelt, 2024-02-14:
Exit 0 is success, so you are probably good there.
Should be up and running at localhost:2900

Do you have any proxy environment variables set?

mdpalme, 2024-02-14:
> I'm pretty sure podman is bugged on redhat 9.1. Please try on 9.2

I've updated to RHEL 9.2. Looks to have the same problem.

> Exit 0 is success, so you are probably good there. Should be up and running at localhost:2900
> 
> Do you have any proxy environment variables set?

Yes, there are proxy environment variables set:

https_proxy=http://<proxy>:80
HTTPS_PROXY=http://<proxy>:80
HTTP_PROXY=http://<proxy>:80
http_proxy=http://<proxy>:80

I'm able to **successfully run version 5.12.0** on this machine with the same environment.

Also, I just noticed... When I attempt to connect with a browser... I don't get `ERR_CONNECTION_REFUSED` or `PAGE NOT FOUND` I just get a blank webpage... It's just white.

Also, in `compose.yaml`  I've replaced all of the instances of `user: "${OPENC3_USER_ID:-1001}:${OPENC3_GROUP_ID:-1001}"` with `user: 0:0`

jmthomas, 2024-02-15:
The traefik line doesn't look good but the rest looks nominal.

What browser version are you using? If you open your browser console you might see some javascript errors. If so you probably need a more modern browser.

mdpalme, 2024-02-16:
Chrome - Version 121.0.6167.185 (Official Build) (64-bit)
Chrome - Version 121.0.6167.161 (Official Build) (64-bit)
These seem to perform fine when COSMOS 5.14.2 is running from Docker/Windows

Microsoft Edge - Version 121.0.2277.112 (Official build) (64-bit)
Same problem

Firefox - 115.7.0esr (64-bit)
Same problem

[image]

ryanmelt, 2024-02-16:
You appear to be using the hostname (ascportal).   We have content security policy configured by default to not allow access to names besides localhost without using https.

You can either setup SSL, or switch traefik to the allow-http configuration file, and then you should be good to go.   See the readme at https://github.com/openc3/cosmos-project for directions.

mdpalme, 2024-02-17:
That seems to do the trick! Thank you!

closed 2024-02-17 by mdpalme