OpenC3/cosmos #1124: COSMOS is unable to find Python read conversion file when building

mkalin2, 2024-02-28:
I have a Python read conversion file in my TARGET/lib directory that was generated using the conversion code generator. 

When I go to build my gem file I get the following error:
```
RuntimeError : Error processing /tmp/d20240228-10-jpy1m2/ENCODER/cmd_tlm/tlm.txt:

tlm.txt:20: READ_CONVERSION encoder_extract_id_conversion.py TELEM_ID_PART_NUM
Error: Unable to require encoder_extract_id_conversion.py due to cannot load such file -- encoder_extract_id_conversion.py. Ensure encoder_extract_id_conversion.py is in the OpenC3 lib directory.
Backtrace:
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet_config.rb:488:in `rescue in process_current_item'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet_config.rb:478:in `process_current_item'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet_config.rb:232:in `block in process_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/config/config_parser.rb:217:in `parse_loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/config/config_parser.rb:217:in `parse_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/packets/packet_config.rb:139:in `process_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/system/system.rb:140:in `block in add_target'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/system/system.rb:139:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/system/system.rb:139:in `add_target'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/system/system.rb:128:in `block in initialize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/system/system.rb:128:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/system/system.rb:128:in `initialize'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/models/target_model.rb:608:in `new'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/models/target_model.rb:608:in `deploy'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/models/plugin_model.rb:233:in `block in install_phase2'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/config/config_parser.rb:217:in `parse_loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/config/config_parser.rb:217:in `parse_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/openc3/models/plugin_model.rb:226:in `install_phase2'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/../bin/openc3cli:313:in `validate_plugin'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.13.0/lib/../bin/openc3cli:644:in `<main>'
```

However, if I change the read conversion files to be Ruby files and update the tlm.txt file accordingly. This error goes away and COSMOS is able to find the files. Is this a bug or do I need to do something different for Python conversion files?

jmthomas, 2024-03-01:
You probably didn't create your target with the `--python` option so now it's a Ruby target (which is also the default). Open the `ENCODER/target.txt` and add the following to the top:
```
LANGUAGE python
```
If you already see `LANGUAGE ruby` then obviously change it to `LANGUAGE python`. That should fix it!

mkalin2, 2024-03-05:
The target was created as a python target and has
``` 
LANGUAGE python
```
at the top of the file.

When I go to build the plugin the error message includes the following path:
`RuntimeError : Error processing /tmp/d20240305-10-bkkmxk/ENCODER/cmd_tlm/tlm.txt:`

I think this expected, but how can I check what is getting copied into that tmp directory upon building?

jmthomas, 2024-04-16:
@mkalin2 I'm unable to reproduce on latest. Can you reproduce? If not I think this might be OBE.

mkalin2, 2024-04-17:
Yep issue seems to have gone away!

closed 2024-04-17 by jmthomas