OpenC3/cosmos #1128: Unable to use custom protocol

Lakshmi5785-dev, 2024-03-04:
I am trying to use cutom protocol by following the below steps as describet in the documetation.

1. created file my_read_protocol.rb , this file is under openc3-cosmos-init/plugins/packages/opec3-cosmos-TARGET1/targets\TARGET1\lib\my_read_protocol.rb

[image]

3. Added the below line in my plugin.txt which is at openc3-cosmos-init\plugins\packages\openc3-cosmos-TARGET1/
   PROTOCOL READ MyReadProtocol

4. Created and loaded the gemfile.
  But I am unable to see the print "For Testing " in the logs. 
The read_data is getting invoked from the main protocol.rb instead of my_read_protocol.rb

I have tried using my_reject_protocol as well,but it didn't work. Am I missing something here?
Please suggest

jmthomas, 2024-03-04:
Please post your full plugin.txt

Lakshmi5785-dev, 2024-03-04:
Below is my plugin.txt

VARIABLE target_name TARGET1

TARGET BLOCKONE <%= target %>
INTERFACE <%= target %>_INT tcpip_client_interface.rb host.docker.internal 8080 8080 10.0 nil LENGTH 0 32 4 1 LITTLE_ENDIAN
  MAP_TARGET <%= target %>

  PROTOCOL READ my_read_protocol.rb

Lakshmi5785-dev, 2024-03-04:
And also I can see the below error from the logs is
 "message":"Unable to require my_read_protocol.rb due to cannot load such file -- fmy_read_protocol.rb. Ensure file_download_protocol.rb is in the OpenC3 lib directory.","type":"log"}

I have placed my_read_protocol.rb in the target\lib directory example openc3\openc3-cosmos-init\plugins\packages\openc3-cosmos-demo\targets\INST\lib - here

ryanmelt, 2024-03-04:
A few issues:
1. You shouldn't be building your own plugins inside our openc3-cosmos-init directory structure.  That is for our init container.  Create your own plugin folder somewhere else standalone using our generators: https://docs.openc3.com/docs/getting-started/generators
2. target_name isn't an allowed variable name, so I'm surprised it isn't giving you errors. target_name and microservice_name are reserved variable names.  Needs to be something unique like my_target_name.
3. Your protocol my_read_protocol.rb needs to be in a lib folder from the same target it is used in.  Otherwise you will have to reference it with the given target name.   I recommend putting protocols in a top-level plugin lib folder rather than target/lib as that will remove a lot of pathing problems.

Lakshmi5785-dev, 2024-03-04:
Thank you Reyon.
Point1 I will try this
2. The variable name is working fine for me not throughing any error.
3. I am following the same folder structre as followed for the my_reject_protocol.rb

closed 2024-03-13 by jmthomas

Lakshmi5785-dev, 2024-03-13:
I am unable to use puts or Logger.info inside the custom protocol(If I use them I am unable to connect to the target).
Is there a way to debug the custom protocol.

ryanmelt, 2024-03-13:
Both puts and Logger are available in protocols.  Puts will come out in the operator container logs.

Shivaraj199, 2024-06-18:
hello @ryanmelt 
 I am running the docker on linux and I'm having trouble sending CAN messages using a custom interface. I am using the project from https://github.com/OpenC3/cosmos
I am geting the Error sending TEST_CAN SEND_MESSAGE due to RuntimeError: Interface not connected: TEST_CAN_INT

I have attached plugin.txt file

VARIABLE test_can_target_name TEST_CAN
TARGET TEST_CAN <%= test_can_target_name %>
INTERFACE <%= test_can_target_name %>_INT lib/can_interface.rb can0
MAP_TARGET <%= test_can_target_name %>

Thank you

ryanmelt, 2024-06-18:
@Shivaraj199 Don't add questions on old tickets.  Open a new one.